RPi-Distro/repo #176: Default screensaver m6502 uses 100% CPU

PaulKGrimes, 2020-06-02:
The X screensaver `m6502` consumes 100% CPU if it is the randomly chosen screensaver selected when the screensaver is active.

It should probably be disabled by default instead of being one of the options that could be chosen at random when the screensaver is active.

XECDesign, 2020-06-03:
xscreensaver doesn't come from our repo, we don't pre-install or recommend it, so this probably isn't the place to report it.

Looks like the issue has been raised before, but had no activity in years:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=851718

I'll leave it open for now and take a look when I have a spare moment.

PaulKGrimes, 2020-06-03:
I'm now doubting my writing of xscreensaver above: it's whichever screensaver is enabled by default on installing the raspberrypi-ui-mods package (I started from a lite install), and I think should be the same as that is enabled by default when installing Raspberry Pi OS.

Thanks for looking.

XECDesign, 2020-06-03:
It would be xscreensaver, but in your case it's getting dragged in as a recommended dependency of something. By default, we don't include any screensaver.

closed 2020-12-08 by XECDesign