RPi-Distro/repo #178: rp-bookshelf package metadata has 404 homepage URL

trejan, 2020-06-07:
rp-bookshelf lists https://github.com/raspberrypi-ui/rp_bookshelf as homepage but that is a 404.

> Package: rp-bookshelf
> Version: 0.4
> Architecture: armhf
> Maintainer: Simon Long
> Installed-Size: 170
> Depends: libc6 (>= 2.4), libcurl3-gnutls (>= 7.16.2), libgdk-pixbuf2.0-0 (>= 2.22.0), libglib2.0-0 (>= 2.35.9), libgtk2.0-0 (>= 2.18.0), libcurl4
> **Homepage: https://github.com/raspberrypi-ui/rp_bookshelf**
> Priority: optional
> Section: admin
> Filename: pool/main/r/rp-bookshelf/rp-bookshelf_0.4_armhf.deb
> Size: 40308
> SHA256: 20d0bbc984cd7b8ef6b67a41e3c7bddb868546f41bd9b7df4cf16750e138cf20
> SHA1: 7bcd129e672736b10936929275378bf7e6a1e829
> MD5sum: fcae7944774da57996ca81947095bbb5
> Description: Raspberry Pi publications
>  Browser for Raspberry Pi Press publications in PDF format
> Description-md5: 3e3091d8753bf26fb8674807a4641d7c
>

trejan, 2020-06-07:
piwiz also has the same problem with a homepage of https://github.com/raspberrypi-ui/piwiz

Have both of these been removed from Github? The source for both is available via the apt source still.

XECDesign, 2020-06-11:
@spl237

spl237, 2020-06-11:
Our GitHub repos are frequently made private, for various reasons. Source code is always available by the recommended method, which is via apt source.

closed 2020-06-11 by spl237

kneekoo, 2020-09-05:
> Our GitHub repos are frequently made private, for various reasons. Source code is always available by the recommended method, which is via apt source.

But then how can we report issues, offer feedback or contributions? Do we submit an issue here?

spl237, 2020-09-07:
> > Our GitHub repos are frequently made private, for various reasons. Source code is always available by the recommended method, which is via apt source.
> 
> But then how can we report issues, offer feedback or contributions? Do we submit an issue here?

Wait for the repo in question to be made public again.

kneekoo, 2020-09-07:
That's an unproductive way to handle bug reports. Who's going to sit on those reports and refresh your GitHub page on a daily/weekly basis until the repository might become public?