RPi-Distro/repo #179: zstd causes Bus error

jfikar, 2020-06-14:
There is a problem with zstd package in recent Raspberry Pi OS (armhf). When trying to compress anything with zstd, it generates a Bus error and stops. The version is zstd-1.3.8+dfsg-3+rpi1. It worked in the past. I'm using RPi4.

```
dd if=/dev/urandom bs=1M count=10 | zstd > test.zst 
Bus error
```

Debian 10.4 zstd-1.3.8+dfsg-3 works fine both armhf and aarch64. Experimental Raspberry Pi OS aarch64 works as well, version zstd-1.3.8+dfsg-3.

jfikar, 2020-06-15:
The Bus error occurs when compression level is between 1 and 4 (default compression level without specifying anything is 3). From 5 up, there is no error. Decompression is also fine.

zstd -3 file.... crash
zstd -5 file.... OK

XECDesign, 2022-01-19:
Seems to be resolved in Bullseye?

jfikar, 2022-01-20:
I've just checked, Bullseye has the version 1.4.8 of zstd and the problem is gone.

XECDesign, 2022-01-20:
Many thanks for confirming.

closed 2022-01-20 by XECDesign