RPi-Distro/repo #182: mosquitto-clients : Depends: libmosquitto1 (= 1.5.7-1) but 1.5.7-1+deb10u1 is to be installed

skullydazed, 2020-07-13:
It looks like libmosquitto1's version was incremented but mosquitto-clients Depends was not bumped to match. This makes it impossible to install `mosquitto_pub` and `mosquitto_sub` for IoT use.

```
pi@hwy36:~ $ sudo apt install mosquitto-clients
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto-clients : Depends: libmosquitto1 (= 1.5.7-1) but 1.5.7-1+deb10u1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

skullydazed, 2020-07-13:
I've discovered a workaround until the dependencies can be corrected:

```
apt download mosquitto-clients
sudo dpkg --force-all -i mosquitto-clients_1.5.7-1_armhf.deb
```

skullydazed, 2020-07-14:
I have a better workaround in the form of a shim package that Provides: libmosquitto1. This leaves apt in a working state. Simply install the attached shim package (or use the .equivs file to build your own) and after that you can `sudo apt instal mosquitto-clients`.

[libmosquitto1-shim.zip](https://github.com/RPi-Distro/repo/files/4915854/libmosquitto1-shim.zip)

XECDesign, 2020-10-02:
Is this still an issue? Everything looks okay on this end?

closed 2020-12-08 by XECDesign