RPi-Distro/repo #183: apt upgrade fails on /boot being mounted ro.

gmos, 2020-07-20:
If apt upgrade wants to upgrade a bootloader or kernel on /boot it fails because /boot is normally mounted ro.
It is easily fixable by remounting /boot rw and then do an apt --fix-broken install, but it should not fail during the upgrade.apt upgrade
This happened last with upgrade from kernel 1.20200601-1 to kernel 1.20200717-1. But also with a previous upgrade.

Some info:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

This is on a Pi Zero W.

XECDesign, 2020-07-20:
> /boot is normally mounted ro.

It certainly shouldn't be. I don't know what the expected behaviour would be if you're trying to write files to a read only file system, but failing seems like the right thing to do.

gmos, 2020-07-21:
Indeed.  So the bug is in apt or in the scripting of the kernel/bootloader upgrade.  Apt/script should remount /boot in rw mode before the actual upgrade starts.  As it did in the past.

Kind regards,
Gijs Mos
M: +31 653 939 920
(mobile device)

XECDesign, 2020-07-21:
By default, /boot is not mounted read only though. If the user has decided to make it read only for some reason, it's not up to a package maintainer to mess with that.

gmos, 2020-07-21:
Then that may be the problem.  I’m pretty sure I have started this card with a vanilla install of Raspbian Buster Lite a month or two ago.  And that I never have changed the mounting mode of the boot partition myself.
But I’ll have to set-up an SD card for a new Pi anyway.  So I’ll get the latest official Buster Lite image an report back here how /boot ends up after a fresh install.

closed 2020-12-22 by no-response[bot]