RPi-Distro/repo #185: [Information Disclosure] OpenSSH_7.4p1 Raspbian-10+deb9u7 discloses OS version

PinkDev1, 2020-07-21:
OS: Raspbian GNU/Linux 9.11 (stretch) armv7l (output of `uname -r` is _Raspbian: 4.19.66-v7+_)
Model: Raspberry Pi 3 Model B Rev 1.2
Kernel: 4.19.66-v7+

The Raspbian-10+deb9u7 release of OpenSSH_7.4p1 sends over the "Raspbian-10+deb9u7" text when communicating SSHD version to a client.
This is considered an Information Disclosure error, because SSHD shouldn't disclose OS Version information to clients.

To verify this is true: Run CrackMapExec against OpenSSH_7.4p1 Raspbian-10+deb9u7 with a command like the following:

./cme --verbose ssh -u pi --port 2322 192.168.0.10
(I changed the default SSHd port from 22 to 2322)
CrackMapExec(github.com/byt3bl33d3r/CrackMapExec) uses the paramiko library(github.com/paramiko/paramiko) to dectect SSH version.

If you traceback the output of CME, you'll find that it's just paramiko "reading a line from the socket" and parsing it to get the version information.

[image]

I reported the bug to the [OpenSSH Bug tracker](https://bugzilla.mindrot.org) ([Bug 3196](https://bugzilla.mindrot.org/show_bug.cgi?id=3196)), but it got closed as RESOLVED INVALID with the following note: 

`That's something added by the OS vendor, either in code or via the VersionAddendum option in sshd_config.  It's not something we have any control over.  You will need to take it up with them.`

I have checked the _VersionAddendum_ option in my sshd_config, and it was already set to none (Yes, uncommenting changes nothing, as it's already using the default value):
`#VersionAddendum none`

So, here we are. Why does this specific release of SSHd do this, and is there a way to prevent it?

XECDesign, 2020-07-22:
This repo is for archive.raspberrypi.org package.

Raspbian bugs should be reported here:

https://bugs.launchpad.net/raspbian/+bugs

It may be worth checking whether Debian does the same thing, because in that case, that would be where the report should go.

PinkDev1, 2020-07-22:
Roge you that.
Bug opened on https://bugs.launchpad.net/raspbian/+bug/1888576

closed 2020-07-22 by PinkDev1