RPi-Distro/repo #186: VNC has issues when you do not renew/change the screen resolution

rugk, 2020-07-23:
We have VNC support (ref. https://github.com/raspberrypi/noobs/issues/116) and [even docs for it](https://www.raspberrypi.org/documentation/remote-access/vnc/), solving [an old question](https://raspberrypi.stackexchange.com/a/114572/119283), but there seems to be a big issue with the system resolution:

If you go through the setup as intended (in my case only after enabling direct capture mode and rebooting, though that may be unrelated) you need to **go to raspi-config and reset the screen resolution to something non-default**.
I seem to be not the only one experiencing this as that workaround has been described here:
https://desertbot.io/blog/headless-raspberry-pi-4-remote-desktop-vnc-setup#step-11-important-change-the-default-screen-resolution

Though as no issue has been reported for this, I'd guess I'll open one here. You can find the full STRs in the blog post there.

----

moved from https://github.com/raspberrypi/noobs/issues/599

lurch, 2020-07-23:
@rguk You appear to be confusing NOOBS (the installer / bootloader) and Raspberry Pi OS (the actual Operating System loaded by NOOBS). They both have their own VNC options, but they're actually quite different things.

rugk, 2020-07-28:
The NOBS people told me the same… :roll_eyes: 

Whatever it is, this issue is about the **Raspberry Pi OS**, i.e. about VNC'ing to the _installed_ version…

XECDesign, 2021-08-31:
AFAIK, the issue is that if you don't have a display plugged in, no framebuffer device is created, which is the intended behaviour, so there's nothing for VNC to display. Selecting another resolution, removes HDMI detection and created the framebuffer anyway.

I guess the timing was a bit unfortunate due to the overhaul of the Documentation repo, but I believe that was the correct place for it. Sorry you're getting bounced around between repos without much luck.

closed 2021-08-31 by XECDesign