RPi-Distro/repo #188: v4l2loopback-dkms cannot be installed or upgraded

vincent-olivert-riera, 2020-07-23:
The `v4l2loopback-dkms` cannot be installed or upgraded because the kernel module fails to compile against one of the versions. Look:

```
# apt install v4l2loopback-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l2loopback-dkms is already the newest version (0.12.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up v4l2loopback-dkms (0.12.1-1) ...
Removing old v4l2loopback-0.12.1 DKMS files...

------------------------------
Deleting module version: 0.12.1
completely from the DKMS tree.
------------------------------
Done.
Loading new v4l2loopback-0.12.1 DKMS files...
It is likely that 5.4.51-v7l+ belongs to a chroot's host
Building for 5.4.51+, 5.4.51-v7+, 5.4.51-v7l+ and 5.4.51-v8+
Building initial module for 5.4.51+
Error! Bad return status for module build on kernel: 5.4.51+ (armv7l)
Consult /var/lib/dkms/v4l2loopback/0.12.1/build/make.log for more information.
dpkg: error processing package v4l2loopback-dkms (--configure):
 installed v4l2loopback-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

It is actually caused by a warning being treated as an error:

```
# cat /var/lib/dkms/v4l2loopback/0.12.1/build/make.log
DKMS make.log for v4l2loopback-0.12.1 for kernel 5.4.51+ (armv7l)
Thu 23 Jul 19:00:04 BST 2020
Building v4l2-loopback driver...
make -C /lib/modules/5.4.51+/build M=/var/lib/dkms/v4l2loopback/0.12.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c: In function ‘vidioc_qbuf’:
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c:1509:4: error: implicit declaration of function ‘v4l2_get_timestamp’; did you mean ‘v4l2_get_subdevdata’? [-Werror=implicit-function-declaration]
    v4l2_get_timestamp(&b->buffer.timestamp);
    ^~~~~~~~~~~~~~~~~~
    v4l2_get_subdevdata
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1709: /var/lib/dkms/v4l2loopback/0.12.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51+'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

vincent-olivert-riera, 2020-07-24:
`v4l2loopback-dkms` does not build against 5.1.x kernels. The issue has been fixed upstream:
https://github.com/umlaeute/v4l2loopback/pull/219/commits/0b8feb80fdef9a415d8250bca1790b3ff23e8391

Would it be possible to get this package updated?

isolin, 2020-07-26:
My video streaming setup stopped working after updating the kernel, due to this issue :(

nschimme, 2020-08-16:
In order to work around this:

```
wget http://ftp.us.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb
sudo apt install ./v4l2loopback-dkms_0.12.5-1_all.deb
```

jlg89, 2020-11-10:
This is what I get:

$ uname -r
5.4.51-v7+

$ sudo apt install ./v4l2loopback-dkms_0.12.5-1_all.deb

Unpacking v4l2loopback-dkms (0.12.5-1) over (0.12.5-1) ...
Setting up v4l2loopback-dkms (0.12.5-1) ...
Loading new v4l2loopback-0.12.5 DKMS files...
It is likely that 5.4.51-v7+ belongs to a chroot's host
Building for 5.4.51+, 5.4.51-v7+, 5.4.51-v7l+, 5.4.51-v8+, 5.4.72+, 5.4.72-v7+ and 5.4.72-v7l+
Module build for kernel 5.4.51+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.4.51-v7+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.4.51-v7l+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.4.51-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Building initial module for 5.4.72+
Done.

v4l2loopback.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72+/extra/

depmod...

DKMS: install completed.
Building initial module for 5.4.72-v7+
Done.

v4l2loopback.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72-v7+/extra/

depmod...

DKMS: install completed.
Building initial module for 5.4.72-v7l+
Done.

v4l2loopback.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72-v7l+/extra/

depmod...

$ sudo modprobe -r v4l2loopback
modprobe: FATAL: Module v4l2loopback not found.

jlg89, 2020-11-10:
I had to run:

\$ sudo apt full-upgrade

Reboot, then:

\$ sudo apt reinstall ./v4l2loopback-dkms_0.12.5-1_all.deb

XECDesign, 2020-11-16:
I've added the backported 5.4-compatible package to our repo. It should be live within an hour.

closed 2020-11-16 by XECDesign

surak, 2022-06-28:
Will this exist for the recent kernels and 64 bit?

RuroWork, 2024-10-21:
sudo apt install linux-headers-$(uname -r) 
wget http://ftp.us.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.13.2-1_all.deb
sudo dpkg -i v4l2loopback-dkms_0.13.2-1_all.deb

XECDesign, 2024-10-21:
> sudo apt install linux-headers-$(uname -r) wget http://ftp.us.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.13.2-1_all.deb sudo dpkg -i v4l2loopback-dkms_0.13.2-1_all.deb

That's definitely something people shouldn't do and won't work.

ioaniatr, 2024-12-22:
For me run teh following solved the issue, as mentioned: [here](https://stackoverflow.com/questions/50361990/how-to-solve-kernel-configuration-is-invalid-issues)
`sudo apt install --reinstall linux-headers-$(uname -r)`

thekidconnect, 2025-04-22:
> For me, run the following solved the issue, as mentioned: [here](https://stackoverflow.com/questions/50361990/how-to-solve-kernel-configuration-is-invalid-issues) `sudo apt install --reinstall linux-headers-$(uname -r)`

When I run this I get the following error output:

```
Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.8.0-57-generic KERNEL_DIR=/lib/modules/6.8.0-57-generic
/build all...(bad exit status: 2)
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/v4l2loopback-dk
ms.0.crash'
Error! Bad return status for module build on kernel: 6.8.0-57-generic (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.7/build/make.log for more information.
   ...done.
Errors were encountered while processing:
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1) 
```

Any known fix for this issue?

XECDesign, 2025-04-23:
I'm guessing google is linking to this issue when they look up generic v4l2loopback-dkms build issues. We're getting a mix of bad advice that doesn't apply to this specific issue and users running completely different systems.

The issue OP reported has been resolved almost 5 years ago. If there's another problem, please open a new issue for it.