RPi-Distro/repo #190: lirc no longer works with GPIO based IR receivers

nickrbogdanov, 2020-08-10:
See https://raspberrypi.stackexchange.com/questions/104008/lirc-irrecord-wont-record-buster-mode2-works

Switching to the `gpio_ir` driver changed the format of the IR data received from the kernel, so everyone has to manually patch and rebuild their lirc package.  Additionally, since there are now separate /dev nodes for RX and TX, people are running two separate lirc instances that now have to listen on two separate sockets.

Short term, Raspberry Pi OS ought to bundle the patch for the data format change, so that lirc works "out of the box".  Longer term, we'd ideally get this fixed upstream in lirc.

Kixunil, 2020-11-07:
I found the root cause of this. This bug was reported and fixed before, but the patch got deactivated in the last version. My understanding is that this patch is not needed/causes problems in **Debian**, but is needed in **Raspbian**. (The changelog doesn't reference any bug, so IDK, you have to ask Andreas Beckmann - anbe@debian.org)

It seems that some sript or whatever blindly pulled that package from Debian ignoring that. The package version was marked `~deb10u1` which I suppose was meant to say it's not intended for any other distribution. So the tooling that pulls packages should be updated to filter-out these.

Building the package is a bit simpler now, you only need to `apt source lirc && apt build-dep lirc && cd lirc-0.10.1 && sed -i 's/^#//' && dpkg-buildpackage`. I'm actually doing it now and I plan to publish my package until this gets resolved, so that people will not have to rebuild.

XECDesign, 2020-11-09:
Thank you both. I've re-enabled that patch and uploaded the updated package, which should be public within an hour.

Kixunil, 2020-11-09:
Great! LOL, I forgot to post the link to my build but it doesn't matter anymore.

closed 2020-11-16 by XECDesign