RPi-Distro/repo #191: mpd: relative backwards seek (rewind function) doesn't work

schlizbaeda, 2020-08-16:
The binary file of **mpd** which is shipped by the Raspberry Pi OS distro performs relative fast forward seeks correctly but it doesn't perform relative rewind seeks:
```
mpc seek +5 # works correctly: Jumps 5 seconds forward inside the current audio track
mpc seek -5 # doesn't jump backwards. Only a short interruption of music occurs
```

I have found this behaviour on the current 32bit version of the OS and on the inofficial 64bit version.

When downloading and compiling the source code of mpd[ as described here](https://www.musicpd.org/doc/html/user.html#compiling-from-source) the rewind feature will work correctly on the Raspberry Pi SBC. A freshly installed image of Raspberry Pi OS should provide a corrected version of mpd anyway.

schlizbaeda, 2020-08-16:
This issue is also described in German language on the German Raspberry Pi forum at
https://forum-raspberrypi.de/forum/thread/45381-schneller-vor-ruecklauf-durch-when-held-auf-next-prev-buttons-buttons-beim-strea/?postID=435914#post435914

XECDesign, 2022-01-19:
Is this still an issue in Bullseye?

closed 2022-02-02 by no-response[bot]

douwe2, 2022-10-20:
I have this issue. WHat information do you need ?

XECDesign, 2022-10-20:
> I have this issue. WHat information do you need ?

I haven't set up mpd in ages and it doesn't seem to just work out of the box, so the minimum set of steps required to reproduce the problem starting with a clean image would be helpful.

douwe2, 2022-10-23:
install latest Raspberry pi OS lite.

sudo apt get install mpd

add a number of wav files to mpd,  ie mpc add /

then seek forward 10 minutes or (so): mpc seek +00:10:00

then seek backwards one minute (for example):

mpc seek -00:01:00

It will then not properly go backward.

But on reflection a better way to go might be to update the rasp os to 
the latest version.

if I recompile mpd on the same raspberry system going backwards works fine.

XECDesign, 2022-10-26:
Seems to work okay. I've tried HMDI and analogue output.