RPi-Distro/repo #194: PcManFM very often hang/crash when using smb network share

beta-tester, 2020-08-20:
hello,
on my RPi4B (4GB) with original RPi4 power supply and stock Pasberry Pi OS (2020-05-27-raspios-buster-armhf.zip),
when i use pcmanfm gets stuck after a while during use of smb network shares.
[image]

it happens to connections with different NAS devices and different samba shares on diferent RPi installations (LibreELEC, Raspberry Pi OS).

for example:
1. open pcmanfm (File Manager)
2. from there, select some files from an attached USB3 stick, right-click, and choose "Cut"
3. in the addressbar, type the uri to the smb network share
e.g.: smb://192.168.1.6/share
4. enter the credentials
(in my case i have no user/password defined for the network shares - guest)
5. on the network share in pcmanfm i created an folder by right-click "New Folder".
6. double click to the new created folder to changed to the folder
7. ricgt-click "Paste" to past the previously "cutted" files.
8. click to "Return to previous folder in history"
9. right-click to new folder and chose "Rename" and give the folder a new name.
10. right-click to new renamed folder and chose "Rename" and give the folder a new name again.
11. try to copy that new renamed renamed folder to another location.
12. change the sort order by clicking a column in the header of the detailed list view.
...

ok, sometimes pcmanfm hangs for ever already at step 5, sometimes it for ever hangs at later a steps 8, 9, 10.
(sometimes pcmanfm means the file/foldername does not exist after renaming at step 10 or 11)

when pcmanfm hangs, it is not possible to click to anything inside the pcmanfm window (File Manager). it is not possible to open a new pcmanfm window - it does not open or it is not responsible as well.
when i have the dialog, that the window is not responding and choose [Cancel] it does not make pcmanfm responsable again,
when i choose [End Process], pcmanfm closes its window and on the desktop all the desktop icons disappears
(Trash bin ison, .desktop icons, mounted device icons).

when i reopen pcmanfm (File Manager) it does not use the preferences anymore
(e.g. i set to use detailed list view and show hidden files as preference - but after the crash, pcmanfm uses big icon view and hides hidden files instead)
[pcmanfm.conf.zip](https://github.com/RPi-Distro/repo/files/5101127/pcmanfm.conf.zip)

after a reboot 
when i open pcmanfm in a console it does not give any error output. also in the dmesg or /var/log/syslog is no related log entry to see.

BTW: pcmanfm on Debian 10 LXDE Live on a PC (debian-live-10.5.0-amd64-lxde.iso) does not show up this issue i have with pcmanfm on Raspberry Pi OS.

XECDesign, 2020-08-24:
@spl237

spl237, 2020-08-24:
I'm not aware of anything we've changed in PCManFM which could affect how SAMBA shares are accessed - probably worth raising this with the PCManFM / LXDE developers.

I'll see if I can repeat this with our changes rolled out.

beta-tester, 2020-09-08:
hard to reproduce... too diffuse... the hanging seems not to be related to smb only.
just now pcmanfm hangs again with no smb connection involved... i just opened pcmanfm and nothing in the file list was selectable/reacting... and the try to close the pcmanfm window ends up in a "not responding" window... now it is completely gray for already about 10 minutes. while writing this comment on the same RPi by using firefox.

ram and swap are not filled unusually,
```
free
              total        used        free      shared  buff/cache   available
Mem:        3920192      848464     2547792      265396      523936     2642940
Swap:        262140           0      262140
```
no provess is at unusual cpu load.
```
Tasks: 195 total,   1 running, 194 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3.5 us,  3.8 sy,  0.0 ni, 92.6 id,  0.1 wa,  0.0 hi,  0.1 si,  0.0 st
MiB Mem :   3828.3 total,   2477.9 free,    832.9 used,    517.5 buff/cache
MiB Swap:    256.0 total,    256.0 free,      0.0 used.   2575.7 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND    
  896 pi        20   0   98008   7496   5824 S  17.3   0.2 908:23.62 conky      
  710 root      20   0  285556 123860  54948 S   2.3   3.2 144:59.89 Xorg       
  836 pi        20   0  178592  44764  27708 S   0.7   1.1  61:57.13 lxpanel    
 2332 pi        20   0   99412  29776  20200 S   0.7   0.8   4:37.13 pcmanfm    
  632 gpsd      10 -10   22636   3520   3084 S   0.3   0.1  13:41.99 gpsd       
  769 pi        20   0    6772   3796   3092 S   0.3   0.1   0:25.35 dbus-daem+ 
...
```
no unusual log output...

does pcmanfm has a verbose or debug mode where it spits out error/warning messages to see whats going on?

beta-tester, 2021-02-11:
while an `apt upgrade` i saw, that there was a package `pcmanfp-gdb` updated. this gave me the hint to try to attack me with gdb to the pcmanfm process while it is hanging.

today i just got my chance to try it.
pcmanfm hang for ever again. i just opend pcmanfm, to open a subfolder of `/srv` and pcmanfm stoped responding.
no network share involved. nothing special...

so i attached me to the process:
`gdb pcmanmf -p 1168`, where 1168 the PID of the running pcmanfm process is

```
(gdb) tbreak
Temporary breakpoint 1 at 0xb6238cd0: file ../sysdeps/unix/sysv/linux/poll.c, line 29.
(gdb) info break
Num     Type           Disp Enb Address    What
1       breakpoint     del  y   0xb6238cd0 ../sysdeps/unix/sysv/linux/poll.c:29
(gdb) step
[Switching to Thread 0xb52a3210 (LWP 1197)]

Thread 2 "gmain" hit Temporary breakpoint 1, __GI___poll (timeout=4047, nfds=2, fds=0xb49005c8) at ../sysdeps/unix/sysv/linux/poll.c:29
29	in ../sysdeps/unix/sysv/linux/poll.c
(gdb) step
__libc_disable_asynccancel (oldtype=0) at ../nptl/cancellation.c:69
69	../nptl/cancellation.c: No such file or directory.
(gdb) step
72	in ../nptl/cancellation.c
(gdb) step
__aeabi_read_tp () at ../sysdeps/unix/sysv/linux/arm/aeabi_read_tp.S:52
52	../sysdeps/unix/sysv/linux/arm/aeabi_read_tp.S: No such file or directory.
(gdb) step
53	in ../sysdeps/unix/sysv/linux/arm/aeabi_read_tp.S
(gdb) step
__libc_disable_asynccancel (oldtype=<optimized out>) at ../nptl/cancellation.c:75
75	../nptl/cancellation.c: No such file or directory.
(gdb) step
79	in ../nptl/cancellation.c
(gdb) step
81	in ../nptl/cancellation.c
(gdb) step
83	in ../nptl/cancellation.c
(gdb) step
__GI___poll (fds=0xb49005c8, nfds=2, timeout=<optimized out>) at ../sysdeps/unix/sysv/linux/poll.c:29
29	../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
(gdb) step
^C
Thread 1 "pcmanfm" received signal SIGINT, Interrupt.
[Switching to Thread 0xb57e5010 (LWP 1168)]
__GI___poll (timeout=-1, nfds=4, fds=0x1c2c8d0) at ../sysdeps/unix/sysv/linux/poll.c:29
29	in ../sysdeps/unix/sysv/linux/poll.c
(gdb) step
^C
Thread 1 "pcmanfm" received signal SIGINT, Interrupt.
__GI___poll (timeout=-1, nfds=4, fds=0x1c2c8d0) at ../sysdeps/unix/sysv/linux/poll.c:29
29	in ../sysdeps/unix/sysv/linux/poll.c
```

but i am not familary with gdb and don't know what to do with it...
to me it looks like it is polling for something for ever and never get to a point of proper timeout handling...

spl237, 2021-02-11:
I'm afraid attaching a debugger to the file manager and hitting break is unlikely to tell you anything useful! Like most Linux applications, the file manager is multi-threaded, and a random break could end up in any thread; even if an application has hung, then a random break is very unlikely to hit the thread which has hung; it can hit any of the code which is still running. In this case, it looks as if you have hit some generic C library Linux code - it isn't the location of your problem. The only way to get a debugger to give you useful information in a multi-threaded environment is to set breakpoints before you start in the areas of the code where there might be problems.

All that said, I do not believe that what you are seeing is a problem with pcmanfm itself; it looks to me as if there is something about what you are trying to access which prevents pcmanfm seeing the contents of the directory in question. There should not be anything in /srv on a standard Raspbian install; if there is something in there, it suggests you have installed some network server software, which in turn means that the problem may well be that something else on your network is grabbing that directory and preventing pcmanfm from displaying it. Before blaming pcman, I suggest you investigate what other processes are using the directory which you are having trouble accessing, what device it is on (it might be something as simple as a failing SD card), what other devices can access that directory, etc. pcmanfm is a very long established piece of software, and I have not seen other reports of this problem, which suggests it is a problem with your system rather than a pcmanfm bug.

closed 2021-02-11 by spl237

beta-tester, 2021-02-11:
yes, you are right, the folder /srv is filled by me.
/srv/iso
/srv/nfs/...
/srv/tftp/...

a tried to open /srv/iso. in that folder there are iso images.
those iso images are mounted to folders in /srv/nfs/...
it is running as an PXE-server.
how could this affect pcmanfm?

and i have those issues also with my home folder sometimes. and in there is nothing fancy.
maybe the SD-card?... but after pcmanfm is killed by the system, i can use the system as usual. even pcmanfm is working and can access to the popints, where it stopped responding before.
i have tried different SD-cards (no fake sd-cards).

i am out of ideas.. i hoped the gdb gave a hint...
is there an option where i can enable a verbose log to get output from pcmanfm that shows what is going on?

spl237, 2021-02-11:
To be honest, you're asking in the wrong place. PCManFM (and libFM, the underlying library it uses) are third party software that we take from LXDE. I do make a few changes to it, but they are almost entirely confined to the UI and are very unlikely to affect the problem you are seeing. You might have more luck raising an issue on the LXDE site - https://github.com/lxde/libfm - where there has recently been some activity with bug fixes being pushed to the relevant repo.

beta-tester, 2021-03-04:
only an informational comment, in case sombody else observed the same issue...

WORKAROUND:
if PcManFM hangs and you want to prevent PcManFM from dragging the desktop (tis icons/functionality) to the death,
do not click to "End Process" for the PcManFM instance.
instead try to open a second instance of PcManFM.
both instances will close/crash immediately, but the Desktop (and its icons/functionality) keeps intact.