RPi-Distro/repo #195: PcManFM: another %20 issue left

beta-tester, 2020-08-24:
@spl237 / Simon Long,
i found one place left where the %20 issue in PCmanFM is present.
when you copy a file with spaces in its filen ame to a samba share but the file already exists at the destination...
then a dialog pops up for what to do (skip, overrider)...
and there the file name in the file name field is still shown with %20 instead of a space.

its a follow up of the comments here: https://www.raspberrypi.org/blog/a-new-raspbian-update/#comment-1521860

[image]

spl237, 2020-09-01:
I've just tried this with all my changes rolled out, and it still happens, so it looks as if it is an issue with upstream PCManFM or libfm - I'd suggest raising an issue with the original authors, as they are better placed to fix this. If I get a chance I'll see if I can see anything obvious, but it's not likely to be code I've looked at, so it'll be easier if the original authors can investigate.

spl237, 2020-09-01:
I've found a potential fix for this, which you are welcome to try (assuming you are happy to rebuild the relevant code yourself).

In the source code for libfm4, in the file src/gtk/fm-progress-dlg.c, change line 270 from:

     tmp = g_filename_display_name(fm_path_get_basename(path));

to

     tmp = g_filename_display_name(fm_path_display_basename(path));

This fixes the problem for me, but I don't know if it will lead to problems elsewhere, so any testing you can do would be appreciated!

beta-tester, 2020-09-01:
@spl237 , hello Simon...
I beg your pardon for that issue report...

you are right - i am wrong - debian HAS that issue as well.
[image]

[image]

maybe i am blind or damaged...
this issue has to be directed to the original authors.
thank you for investigating your time.

spl237, 2020-09-01:
No problem - but if you get a chance, do try the fix above - I think it should work.

beta-tester, 2020-09-01:
how can i test it... do i have to compile the libfm4 locally or can i download the fixed version somewhere?

spl237, 2020-09-01:
You'll have to build it locally:

apt source libfm
sudo apt build-dep libfm
cd libfm-1.3.1
./configure --prefix=/usr --libdir=/usr/lib/arm-linux-gnueabihf

Make the code change as above and then:

make
sudo make install

beta-tester, 2020-09-01:
i will prepare another RPi for building and testing...
the RPi where i am on right now i will not reboot today.
... i will tell you later the result.

beta-tester, 2020-09-01:
yes, your fix is working...
thank you!

PS: will the `sudo make install` be overwritten when a new version of libfm will be published via regular apt update && apt full-upgrade ... or do i have to uninstall the modified version first?

spl237, 2020-09-01:
The version you have built will be overwritten with the next update from apt.

If you can carry on using it and let me know if you find any problems, that would be great - I think it should be safe to add to the version in apt, but it would be good for it to have some more testing first!

beta-tester, 2020-09-01:
ok, i will carry on with this version and let you know when i see any related issue with it.

beta-tester, 2020-09-04:
today i received an updated libfm and pcmanfm via apt...
it is fixed now in the main stream or wasn't just my local modification been overwritten?

spl237, 2020-09-04:
Fixed in the updated version, which will have overwritten your locally-built version with an official release of exactly the same build...

closed 2020-09-04 by spl237