RPi-Distro/repo #196: PulseAudio only treats all output as Mono on the PI4

athrun, 2020-08-26:
Hi!

By default, PulseAudio on the PI4 treats all analog output as Mono, including all HDMI outputs.
I was able to track down the issue and discovered it is simply a configuration file issue that has been [fixed upstream two years ago](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/commit/3455d62e4906174e83bf0ea9af8b584808d3e779).

The PulseAudio version included in the Pi OS is a bit old, but I was able confirm that simply updating the configuration file without upgrading the PulseAudio version will address the problem.

In `/usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf`, simply comment out the `analog-mono` section. This will allow the existing `stereo-fallback` section to take over.

Comment out this section:
```
#[Mapping analog-mono]
#device-strings = hw:%f
#channel-map = mono
#paths-output = analog-output analog-output-lineout analog-output-speaker analog-output-headphones analog-output-headphones-2 analog-output-mono
#paths-input = analog-input-front-mic analog-input-rear-mic analog-input-internal-mic analog-input-dock-mic analog-input analog-input-mic analog-input-linein analog-input-aux analog-input-video analog-input-tvtuner analog-input-fm analog-input-mic-line analog-input-headset-mic
#priority = 7
```

I think including this simple change would make a lot of people using their RPi as a workstation connected to an HDMI monitor happy!

### Before the change:

Notice how both PulseAudio sinks (headphones and HDMI1) are configured as Mono ouputs:

```pi@pi ~> pactl list sinks
Sink #0
	State: IDLE
	Name: alsa_output.platform-bcm2835_audio.analog-mono
	Description: Built-in Audio Analog Mono
	Driver: module-alsa-card.c
	Sample Specification: s16le 1ch 44100Hz
	Channel Map: mono
	Owner Module: 6
	Mute: no
	Volume: mono: 65536 / 100% / 0.00 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.platform-bcm2835_audio.analog-mono.monitor
	Latency: 94282 usec, configured 99954 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 HDMI 1"
		alsa.id = "bcm2835 HDMI 1"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "bcm2835 HDMI 1"
		alsa.long_card_name = "bcm2835 HDMI 1"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-bcm2835_audio"
		sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
		device.form_factor = "internal"
		device.string = "hw:0"
		device.buffering.buffer_size = "8816"
		device.buffering.fragment_size = "2208"
		device.access_mode = "mmap"
		device.profile.name = "analog-mono"
		device.profile.description = "Analog Mono"
		device.description = "Built-in Audio Analog Mono"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output: Analog Output (priority: 9900)
	Active Port: analog-output
	Formats:
		pcm

Sink #1
	State: IDLE
	Name: alsa_output.platform-bcm2835_audio.analog-mono.2
	Description: Built-in Audio Analog Mono
	Driver: module-alsa-card.c
	Sample Specification: s16le 1ch 44100Hz
	Channel Map: mono
	Owner Module: 7
	Mute: no
	Volume: mono: 57371 /  88% / -3.47 dB
	        balance 0.00
	Base Volume: 56210 /  86% / -4.00 dB
	Monitor Source: alsa_output.platform-bcm2835_audio.analog-mono.2.monitor
	Latency: 100645 usec, configured 99954 usec
	Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 Headphones"
		alsa.id = "bcm2835 Headphones"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "bcm2835 Headphones"
		alsa.long_card_name = "bcm2835 Headphones"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-bcm2835_audio"
		sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card1"
		device.form_factor = "internal"
		device.string = "hw:1"
		device.buffering.buffer_size = "8816"
		device.buffering.fragment_size = "2208"
		device.access_mode = "mmap"
		device.profile.name = "analog-mono"
		device.profile.description = "Analog Mono"
		device.description = "Built-in Audio Analog Mono"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output-headphones: Headphones (priority: 9000)
	Active Port: analog-output-headphones
	Formats:
		pcm
```

### After making the change and running `pulseaudio -k`:

```
pi@pi ~> pactl list sinks
Sink #0
	State: IDLE
	Name: alsa_output.platform-bcm2835_audio.stereo-fallback
	Description: Built-in Audio Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 6
	Mute: no
	Volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: alsa_output.platform-bcm2835_audio.stereo-fallback.monitor
	Latency: 94145 usec, configured 99954 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 HDMI 1"
		alsa.id = "bcm2835 HDMI 1"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "0"
		alsa.card_name = "bcm2835 HDMI 1"
		alsa.long_card_name = "bcm2835 HDMI 1"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-bcm2835_audio"
		sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0"
		device.form_factor = "internal"
		device.string = "hw:0"
		device.buffering.buffer_size = "17632"
		device.buffering.fragment_size = "4416"
		device.access_mode = "mmap"
		device.profile.name = "stereo-fallback"
		device.profile.description = "Stereo"
		device.description = "Built-in Audio Stereo"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output: Analog Output (priority: 9900)
	Active Port: analog-output
	Formats:
		pcm

Sink #1
	State: IDLE
	Name: alsa_output.platform-bcm2835_audio.stereo-fallback.2
	Description: Built-in Audio Stereo
	Driver: module-alsa-card.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 7
	Mute: no
	Volume: front-left: 57365 /  88% / -3.47 dB,   front-right: 57365 /  88% / -3.47 dB
	        balance 0.00
	Base Volume: 56210 /  86% / -4.00 dB
	Monitor Source: alsa_output.platform-bcm2835_audio.stereo-fallback.2.monitor
	Latency: 93900 usec, configured 99954 usec
	Flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY 
	Properties:
		alsa.resolution_bits = "16"
		device.api = "alsa"
		device.class = "sound"
		alsa.class = "generic"
		alsa.subclass = "generic-mix"
		alsa.name = "bcm2835 Headphones"
		alsa.id = "bcm2835 Headphones"
		alsa.subdevice = "0"
		alsa.subdevice_name = "subdevice #0"
		alsa.device = "0"
		alsa.card = "1"
		alsa.card_name = "bcm2835 Headphones"
		alsa.long_card_name = "bcm2835 Headphones"
		alsa.driver_name = "snd_bcm2835"
		device.bus_path = "platform-bcm2835_audio"
		sysfs.path = "/devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card1"
		device.form_factor = "internal"
		device.string = "hw:1"
		device.buffering.buffer_size = "17632"
		device.buffering.fragment_size = "4416"
		device.access_mode = "mmap"
		device.profile.name = "stereo-fallback"
		device.profile.description = "Stereo"
		device.description = "Built-in Audio Stereo"
		alsa.mixer_name = "Broadcom Mixer"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card"
	Ports:
		analog-output-headphones: Headphones (priority: 9000)
	Active Port: analog-output-headphones
	Formats:
		pcm
```

beta-tester, 2020-12-08:
today (2020-12-07) an upgrade of PulseAudio was released that fixed the issue for me.

pulseaudio-utils (12.2-4+deb10u1+rpi2)
pulseaudio-module-bluetooth (12.2-4+deb10u1+rpi2)
pulseaudio (12.2-4+deb10u1+rpi2)

XECDesign, 2020-12-08:
Yup, that update was to fix that issue.

closed 2020-12-22 by no-response[bot]

nmisha, 2021-07-27:
But in original raspberry lite this block uncommented. Why we have the issue?

lackarice, 2022-07-07:
nice, help me fix output my problem , thanks a lot!