RPi-Distro/repo #197: screenfetch showing incorrect shell info and is out of date

CyberManifest, 2020-09-01:
screenfetch package version 3.8.0-8 on Raspberry Pi 3 b+ on Raspbian 10 buster running armv7l Linux 5.4.59-v7+ using zsh gives incorrect shell information (a random number) see: https://i.imgur.com/i19Cxdx.png

This has since been resolved in the latest edition from github source: https://github.com/KittyKatt/screenFetch

This has been tested and verified by downloading https://raw.githubusercontent.com/KittyKatt/screenFetch/master/screenfetch-dev
making it executable and launching:
```
wget https://raw.githubusercontent.com/KittyKatt/screenFetch/master/screenfetch-dev
chmod +x screenfetch-dev
./screenfetch-dev
```
Please see: https://i.imgur.com/UJxlbBL.png
It's presumed the package needs to be updated upstream from Debian 10 Buster armhf

XECDesign, 2020-09-01:
Debian and Raspberry Pi OS are not rolling release distros. Packages get updated between releases, but in general the versions within a release don't change.

Also, screenfetch does not come from our repo, so we don't have control over it, unless we want to take full responsibility for maintaining while Buster is the stable release.

The alternative is to report it as a bug to Debian, then any fixes they apply will end up in Raspberry Pi OS.

closed 2020-09-01 by XECDesign

CyberManifest, 2020-09-01:
I see, so you all don't mind shipping broken packages and don't care for the vetting process. I'm fully aware Debian and Raspberry Pi (Raspbian) are not rolling release distros; I've used both kinds Debian and Arch based. This is why I don't typically bother reporting bugs. It's not generally appreciated. I fixed it on my system so I don't care about this issue anymore if y'all aren't going to care. I took action, if you want an improvement then YOU can report it to Debian upstream. 

> Also, screenfetch does not come from our repo, so we don't have control over it, unless we want to take full responsibility 
for maintaining while Buster is the stable release.

Funny, that's the only repo I have active and I initially installed screenfetch via: `sudo apt install screenfetch`; I have my bash and zsh logs to confirm if you'd like.

Report it to Debian Your Self you Debian Downer.