RPi-Distro/repo #198: wrong keyboard layout in "Welcome to Raspberry Pi" wizard

beta-tester, 2020-09-04:
hello,
this is a issue for a very long time already. [see...](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=243717&start=300#p1671046)
now i will report it this way...

at very first start into the Raspberry Pi OS desktop, the "Welcome to Raspberry Pi" wizard mess up with the keyboard layout settings.
when i setup something different than en_UK, during entering the password there is still the en_UK keyboard layout active.
this may cause login problems the next time when you typed in a password that keys swapped from en_UK to what ever you choosed.

this will be not a good first experience of using a RPi when you cant log in with the password of your choice...

to reproduce:
1. download and write the image `2020-08-20-raspios-buster-armhf.zip` to an SD card.
2. plug the SD card to your RPi (with connected HDMI Monitor/TV and real keyboard with a different kayboard layout than en_UK)
in my case with German de_DE layout - `QWERTZ` instead of `QWERTY` of en_UK.
3. power on you RPi and wait until the `Welcome to Raspberry Pi` wizard appears to the desktop.
4. change the County to `Germany` (language will then automatically change to `German` and timezone to `Berlin`)
5. because i prefere to have english as the display language i ticked the option `[x] Use English language`.
6. then go `[Next]`
7. after `Settin location - please wait` progress message disappear the `change password` dialog shows up.
8. un-tick the option `[_] Hide characters` to see, what you type in as password.
9. when you now type in a password you will see, that the keyboard layout en_UK is still active instead of the the keyboard layout of your choosen Country/Language.

[image]

i typed in `qwertz`, `QWERTZ`,
but `qwerty`, `QWERTY` was taken to the fields
[image]

XECDesign, 2020-09-04:
@spl237

spl237, 2020-09-04:
Getting localisation changes to reload while an application is still running is awkward - what I would like to happen is that when a new language is selected in the wizard, the wizard immediately uses that language for the rest of the pages, but I cannot find any way to get that to happen with the way Linux localisation support works.

It ought to be possible to force reloading of the keymap while the wizard is running; I'll look into this.

beta-tester, 2020-09-04:
i don't know if the installation wizard of ubuntu is open source, but there you can change the language and it takes the changed keyboard layout for further text fields..
 in debian i am not sure if it has a similar wizard for installation...

XECDesign, 2020-09-04:
It's called ubiquity and it is open source. As far as I know, it just runs setxkbmap, maybe with some flags.

beta-tester, 2020-09-04:
thank you for the link. this is helpful.

spl237, 2020-09-04:
There's already code in the wizard which calls setxkbmap after the keyboard is selected, and if you do this with the wizard launched from the terminal, it does indeed load the new keymap immediately as it is supposed to. I think there is something odd about the state in which a completely clean new image starts up which is screwing up the relevant code - I suspect it is that there is no initial keyboard set at all, and so the code which changes from one keyboard to another fails.

I'll have a look - the fact that it does work correctly when called from the terminal on a configured system indicates that the code is doing the right thing; there's just something about the initial state with which it is unhappy. I'm sure it's a relatively easy fix.

beta-tester, 2020-09-04:
here the link from the other apply_keyboard meton in ubiquity
https://github.com/linuxmint/ubiquity/blob/master/bin/ubiquity#L291

spl237, 2020-09-04:
As above - there is already code to do this in the wizard, which works when run in other circumstances - there is just a problem with the initial state which I need to fix. I don't think it's going to be a big problem to fix, and it's going to be easier for me to fix my own code than to try and work out what other Linuxes have done...

beta-tester, 2020-09-05:
i forgot to mention,
that even when the wizard completed sucessfully and the RPi is rebooted the keyboard layout is still en_UK...

follow the steps above...
... and continue with:
10. `[X] This screen shows a black border around the desktop`, `[Next]`
11. Select your wireless network from the list, `[Skip]`
12. Update Software, `[Skip]` or `[Update]`
13. Setup Completed, `[Restart]`

... wait until you are back to the desktop again...

14. open "Accessories|Text Editor" and type in `QWERTZ` but `QWERTY` is shown
[image]

15. open "Accessories|Terminal" and type in `QWERTZ` but `QWERTY` is shown
[image]

you don't need a german keyboard to reproduce that issue,
when you know the keyboard layout and you know, what it should give...
the row just below the row with the numbers
[image]

when you set up everything following step 1 to step 13... and type the characters of the row below the numbers, starting with `Q` to the left, when you get next to the `T` a `Y` on screen then it is not set up to german keyboard layout.

PS. i am using a Logitech K400 keyboard...

spl237, 2020-09-07:
OK, I've looked into this.

The situation is caused by your checking the "use English language" checkbox. If you select Germany as the country and do not check that box, then the German keyboard is loaded. By selecting that checkbox, the system locale and keyboard are both set to English; Germany is used for the timezone only.

The question is what is the correct behaviour here, because I think there is a valid argument that what is currently happening is correct - the assumption is made that someone who wants to use their Pi in English will also want a UK keyboard map. For example, if someone chooses to use their Pi in English in somewhere like Japan where a non-Roman character set would otherwise be used, it would seem to be wrong to have the keyboard producing Kanji or similar in an interface which is entirely in English. So I think the keyboard map needs to match the selected user interface language.

I can see that you might expect the keyboard to be German, but this doesn't really fit with a user interface which is otherwise in English. So I think the current behaviour is correct, and won't be changed; you'll need to go into the Mouse and Keyboard settings dialog once the wizard has completed and select a German keyboard map.

closed 2020-09-07 by spl237

beta-tester, 2020-09-07:
but then i don't understand the option `[_] Use US Keyboard` when the option `[x] Use English language` already overrides the other settings for keyboard layout...
wouldn't it be better to have a list to choose the keyboard layout instead of only a checkbox? or at least show what keyboard layout will be effectivly be loaded/installed...

i choose English as desktop language because it is nicer to otheres to read when making screenshots or search for status/error messages in internet fives much more hits than searching a german language message.

but the keyboard you can buy in every bodys location is the local language one.
so it makes more sense to me to use the keyboard layout of the selected location.
a keyboard is the human-to-computer interface that is used actively by the fingers.
it is very very uncommen that you go to a local store to by a US or UK keyboard in an non-US od non-UK location.
i think even in USA location they are not happy when the get a UK keyboard layout when they select United States as location and `[x] Use English language`...
ok, if they tick that option, then it's more by accident.

beta-tester, 2020-09-07:
but then i don't understand the option [_] Use US Keyboard when the option [x] Use English language already overrides the other settings for keyboard layout...
wouldn't it be better to have a list to choose the keyboard layout instead of only a checkbox? or at least show what keyboard layout will be effectivly be loaded/installed...

i choose English as desktop language because it is nicer to otheres to read for the community when making screenshots or search for status/error messages in internet and forum gives much more hits than searching a german language message.

but the keyboard you can buy in every bodys location is the local language one.
so it makes more sense to me to use the keyboard layout of the selected location.
a keyboard is the human-to-computer interface that is used actively by the fingers.
it is very very uncommen that you go to a local store to by a US or UK keyboard in an non-US od non-UK location.
i think even in USA location they are not happy when the get a UK keyboard layout when they select United States as location and [x] Use English language...
ok, if they tick that option, then it's more by accident.

spl237, 2020-09-07:
The US keyboard is the most widely-available worldwide, and is used in preference to the local keyboard layout by many people in several countries. So it makes sense to have it as an override.

Someone in the US will get English language automatically as a result of selecting the country to be United States; they have no need to tick the English language override.

The whole purpose of the localisation settings on the wizard is to simplify setup for the majority of users; it is not intended to provide every possible alternative configuration. The configuration you want - an English language UI with a German keyboard layout - is not common. So it is not one which is catered for by the wizard - as before, for unusual setups, you will need to use the full set of localisation options which are included in Raspberry Pi Configuration for that purpose.

beta-tester, 2020-09-07:
ok... then i expect too much from that wizard.
i was seeing it as a kind of installation preparation wizard like in ubuntu...