RPi-Distro/repo #199: Wrong language in Pi Wizard....

Tzarls, 2020-09-05:
...well, kind of. When building a custom image using Pi Gen and setting the default language and locale to spanish and Latin America, at first boot the system shows the correct language in its menus, but the Pi Wizard is still in English. Looking at the code there seems to be no actual translation for it, but here:

[https://projects.raspberrypi.org/es-ES/projects/raspberry-pi-getting-started/4](url)
[image]

There appears to be a spanish version of the Wizard. I'd like to have a startup experience as local as possible and having this first time configuration tool in english can be difficult for non english speakers. So, is there a version of the Wizard with translations or will I have to change the code for this?

spl237, 2020-09-06:
There are no translation files at all for the wizard - it will only work in English, and we rely on members of the community to submit translation files. The image above has been faked by the team who created the webpage - there is no Spanish translation available for the wizard, and there won't be until someone who speaks Spanish submits one...

closed 2020-09-06 by spl237

Tzarls, 2020-09-06:
Translation should be done by creating a .po file with translated strings and locale identifiers, right? Since the wizard's repository is not public, where should we submit the translation? Here?

spl237, 2020-09-07:
I've made the piwiz repo public - please add the relevant po file to the po subdirectory there.

Tzarls, 2020-09-08:
In line 193 of the default po file it says:

msgid "Error creating watch\n"

Whats is "watch" in this context? I need to know this for the translation (which I'm working right now).

spl237, 2020-09-09:
It's setting a watch on a GIO channel; something which issues a callback when a condition is met on a channel - see https://developer.gnome.org/glib/stable/glib-IO-Channels.html#g-io-add-watch

Tzarls, 2020-09-14:
Ok, translation is ready. I had no write privilege for the repository so I created a pull request. I hope that's ok.