RPi-Distro/repo #200: rp-bookshelf performance feedback

kneekoo, 2020-09-05:
I just installed a fresh Raspberry Pi OS on a Sandisk Ultra 128GB, for my Pi 4B 2GB. I opened Bookshelf and it needed **7 minutes** to finish grabbing all the covers/thumbnails for the offered publications. This is too much now, and the more issues get added in each category, the longer it will take Bookshelf to finish getting the images. Currently, we have 170 images taking up **44.6 MiB**.

1. Stop using PNG covers, which is not a good choice when you have a large colour palette. JPG with 90% quality and no EXIF should be perfect.

2. The image resolutions vary wildly. They should be limited to the maximum height that would make sense for display in the Bookshelf. As long as the program won't support HiDPI and it only displays the images at 128px high, that's the resolution that the thumbnails should have been sized to.

To check the difference between the original files and those converted, resized and stripped of EXIF, I used this script in ~/.cache/bookshelf (after I installed imagemagik and exiftool, of course):
```
#!/bin/bash
shopt -s nocaseglob
for img in *.{jpg,jpeg,png}
do
    OLD="$img"
    NEW="${OLD%.*}.jpg"
    mv "$OLD" bkp/
    convert "bkp/$OLD" -resize x128 -quality 90 "./$NEW"
    exiftool -overwrite_original -all= "$NEW"
done
```

The resulting size of the 170 jpg files was **1.5 MiB** and Bookshelf seemed to load them faster than before, but then it had to grab the png covers again, because that's what it knows it has to display _(from cat.xml)_. Now if all those images would be properly done directly on the server, they would take a lot less time to download and display.

Please consider doing this on the server side.

XECDesign, 2020-09-06:
@spl237

spl237, 2020-09-06:
We're aware of the issues and are already looking at how to improve performance at the server end.

closed 2020-09-06 by spl237