RPi-Distro/repo #203: Moving RPi desktop packages back to "ui" component

MichaIng, 2020-09-24:
With the Buster suite, the [ui component](https://archive.raspberrypi.org/debian/dists/buster/ui/) has been abandoned and the RPi desktop packages have been moved into the [main component](https://archive.raspberrypi.org/debian/dists/buster/main/) instead.

This makes it difficult for users and distros, which want to use the official RPi firmware/kernel packages as well as a native LXDE desktop.

The LXDE-based RPi desktop is a beauty, no question, and much work has been put inside to make it that beautiful, functional and responsive on RPi, but some of those changes are not well compatible with the native LXDE session/theme, i.e. what you get when simply installing the `lxde` package and running `startx`:
- https://github.com/raspberrypi-ui/pcmanfm-buster/issues/2
- https://github.com/raspberrypi-ui/lxpanel-buster/issues/3
- Probably related but still matter of investigation: https://github.com/RPi-Distro/repo/issues/202

I understood and respect that it is not the aim and would imply (too) much work to make and keep those packages compatible with native LXDE session without breaking the RPi desktop, but wouldn't it make sense to keep them in a separate repository component (as it was until including Raspbian Stretch)? So one can easily decide/switch between the RPi desktop and Debian LXDE, which means that the unmodified ones from Debian/Raspbian repository are used.

Currently a solution is to pin down the conflicting packages via apt preferences, but that leads to a mixture of both package variants when not checking or knowing exactly which ones belong to the RPi desktop theme only and which are part of e.g. the GPU-accelerated X server/driver and/or are dependencies across other parts of the RPi repository. And without constantly checking/maintaining the pins, one might run into new incompatibilities or failing APT upgrades due to newly introduced dependencies, so the whole concept of APT pinning in this regards is quite fragile.

I know that it is likely not your aim to invest much time for users or distros to thankfully use parts of your repository while avoiding other parts, but instead the main aim is to provide a solid repository for the official Raspberry Pi OS images, so I'm not expecting something here. But I didn't want to miss the chance either to at least share this point of view 😉.

XECDesign, 2020-09-24:
Yeah, I had concerns about Raspberry Pi Desktop breaking LXDE and cutting out features which more advanced users might rely on, so that was why the ui component was added.

Without getting into too much detail, it became a bit of a headache and I didn't see many people trying to use stock LXDE instead of RPD, so it didn't seem worth while to maintain the ui component.

I wish RPD was a complete fork with no clashing file names or packages so that LXDE and RPD could co-exist, but unless there's a simple way to automatically sort packages, I think it's something we'll have to live with.

MichaIng, 2020-09-24:
> I wish RPD was a complete fork with no clashing file names or packages so that LXDE and RPD could co-exist

Yes that was a second alternative I was thinking of, it would have other benefits as well. But that means to create something like `openbox-rpd`, `pcmanfm-rpd` as well, or merge everything in a single package and I guess that complicates implementing upstream updates or migrating to new LXDE + Openbox + PCManFM base versions.

> so it didn't seem worth while to maintain the ui component.

I see if this causes additional maintenance effort, especially if we belong to a rare minority to use native LXDE on RPi. And since I found an acceptable workaround currently via apt pinning, it's okay.

XECDesign, 2020-11-24:
Closing, since there's not much I can do about it.

closed 2020-11-24 by XECDesign