RPi-Distro/repo #204: Package libx11-xcb-dev broken due to incorrect dependency version

benjamincburns, 2020-10-02:
When I try to install `libx11-xcb-dev` on the latest version of RPi OS, it fails due to unmet dependencies.

`Depends: libx11-xcb1 (= 2:1.6.7-1) but 2:1.6.7-1+deb10u1 is to be installed.`

I'm guessing that the `libx11-xcb-dev` package's metadata is incorrect.

benjamincburns, 2020-10-02:
I'm guessing that this is a similar issue to #182

XECDesign, 2020-10-02:
Not seeing any such issue on this end.

Did you run `sudo apt update` first? What's the output of `apt policy`?

benjamincburns, 2020-10-02:
Yes, I ran `sudo apt update.` In the end I followed the workaround suggested in #182 (force installing it via `dpkg`), so I'm not sure if the outout of `apt policy` will be all that telling. I'm not near the device at the moment here, but I'll go ahead and report back with that as soon as I am.

benjamincburns, 2020-12-09:
I wound up ditching this project and the rpi with it. Since I haven't seen a wealth of "me toos" on this issue, I'll go ahead and close it.

closed 2020-12-09 by benjamincburns