RPi-Distro/repo #206: Corrupted kernel file on RPI-Zero-WH (file 19-raspberrypi-kernel_1.20201022-1_armhf.deb)

kumters, 2020-10-24:
Hi guys, I have bed news about the latest kernel .deb package for RPI-Zero-WH. I did upgrade of my RPI and it ended with the broken OS. (I tried it on pure/clean 2020-08-20-raspios-buster-armhf-lite.img with the same result).

**Error (after 2 minutes of updating by apt-get update):
https://pastebin.pl/view/fd901d13
**

**System info:
https://pastebin.pl/view/317f8fe5
**

Conclusion: kernel .deb file is corrupted.

Can you please repair it?
Thank you very much
PJ

popcornmix, 2020-10-24:
These messages:
```
[  883.046784] mmc0: timeout waiting for hardware interrupt.
[  883.046812] [345b7937] IRQ< 400 0
[  883.046822] [345b7938] IRQB d6d10154 400
[  883.046831] [345b7938] FCM< d6d100b0 d6d10154
[  883.046838] [345b7939] RSP  c00 0
[  883.046846] [345b793a] FCM> d6d100b0 d6d10154
```
look like you may have sdcard issues on your existing install. Do you have a spare sdcard you can try installing and upgrading on?

kumters, 2020-10-24:
Yes, I have a totally new 32GB Sandisk, I will try it ASAP:)

kumters, 2020-10-24:
Ok reinstalled clean RPI image, did update+upgrade without errors. So it looks like an issue with that old 8GB SDcard. Thank you very much and please close/reject this issue.

closed 2020-10-24 by kumters