RPi-Distro/repo #207: freeradius weird interaction

JiffB, 2020-10-29:
Hi,
I've got a weird interaction into freeradius, following : https://wiki.freeradius.org/guide/Getting-Started, I created this line : 
```testing Cleartext-Password := "password"```
that is authorized : 
```
$ radtest -x -4   testing   password   127.0.0.1   0   testing123
Sent Access-Request Id 210 from 0.0.0.0:36875 to 127.0.0.1:1812 length 77
	User-Name = "testing"
	User-Password = "password"
	NAS-IP-Address = 192.168.1.254
	NAS-Port = 0
	Message-Authenticator = 0x00
	Cleartext-Password = "password"
Received Access-Accept Id 210 from 127.0.0.1:1812 to 127.0.0.1:36875 length 20
```
which is normal.
But when I de-commented one of the test users that are provided, namely 'steve', who has this line : 
```steve  Cleartext-Password := "testing"```
the test result was… weird : 
```
radtest -x  steve  testing  127.0.0.1  10  testing123
Sent Access-Request Id 173 from 0.0.0.0:53328 to 127.0.0.1:1812 length 75
	User-Name = "steve"
	User-Password = "testing"
	NAS-IP-Address = 192.168.1.254
	NAS-Port = 10
	Message-Authenticator = 0x00
	Cleartext-Password = "testing"
Received Access-Reject Id 173 from 127.0.0.1:1812 to 127.0.0.1:53328 length 20
```
Changing 'steve' original password from "testing" to "test" worked : 
```steve  Cleartext-Password := "test"```
with a good result : 
 ```
radtest -x  steve  test  127.0.0.1  10  testing123
Sent Access-Request Id 53 from 0.0.0.0:38595 to 127.0.0.1:1812 length 75
	User-Name = "steve"
	User-Password = "test"
	NAS-IP-Address = 192.168.1.254
	NAS-Port = 10
	Message-Authenticator = 0x00
	Cleartext-Password = "test"
Received Access-Accept Id 53 from 127.0.0.1:1812 to 127.0.0.1:38595 length 20
```
which is very weird.

Note that this interaction doesn't happen under Debian buster amd64.

Jiff

XECDesign, 2022-01-19:
Is this still an issue in Bullseye?

JiffB, 2022-01-19:
Unfortunately, I can't answer to this at this time, as I migrated my Raspberry Pi 4 to arm64 and do not run Freeradius at this time on it (only hostapd).

XECDesign, 2022-01-19:
Alright, sorry it took this long. I'll close it for now, but if the issue comes up again, do shout and I'll get to it quickly.

closed 2022-01-19 by XECDesign