RPi-Distro/repo #208: Ethernet (eth0) didn't coming back to running after manually putting down from ifconfig/ioctl

jarus0, 2020-11-06:
So basically I want to turn on/off wire & wireless network interface from program or terminal.

For disabling/enabling wlan0 network interface I try below cmd that working completely fine.
```
sudo ifconfig wlan0 down
sudo ifconfig wlan0 up
```
But for eth0 same method not working
below "/var/log/syslog" given for `sudo ifconfig eth0 down`
```
Nov  5 13:18:28 raspberrypi dhcpcd[389]: eth0: carrier lost
Nov  5 13:18:28 raspberrypi dhcpcd[389]: eth0: deleting address fe80::9e89:dc86:fac9:a345
Nov  5 13:18:28 raspberrypi dhcpcd[389]: eth0: deleting route to 192.168.1.0/24
Nov  5 13:18:28 raspberrypi dhcpcd[389]: eth0: deleting default route via 192.168.1.1
Nov  5 13:18:30 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
```
syslog for `sudo ifconfig eth0 up`
Nov  5 13:21:29 raspberrypi kernel: [  216.923295] 8021q: adding VLAN 0 to HW filter on device eth0

Probably dhcpcd can't acquire carrier for eth0.

Below "/etc/dhcpcd.conf" file contain give 
```
interface eth0
static ip_address=192.168.1.246/24
static routers=192.168.1.1
static domain_name_servers=8.8.8.8
metric 400
```

and i have not change in "/etc/network/interfaces" file.

XECDesign, 2020-11-06:
If you've not set up `/etc/network/interfaces` accordingly and disable other network managers, then ifupdown isn't the right tool.

Does `sudo ip link set eth0 down` work?

XECDesign, 2020-11-06:
Not having any issues with either `ip link` or `ifconfig` when trying to do the same. I'm also not seeing the message about `VLAN 0`.