RPi-Distro/repo #209: /etc/shadow- isn't update when changing a p/w

JiffB, 2020-11-06:
Hi,

I just had a problem twice in a row, pi user p/w reset or change (usual p/w didn't work in the graphical input) ; after investigation, the backup file `/etc/shadow-` isn't updated after a p/w change :/ (should be the same timestamp as `/etc/shadow`).

Apparently, this weird behavior is quite recent (however IIRC, the 10/29 was also a p/w problem, may be the package date below is more relevant) : 
```
# la /etc/shadow*
-rw-r----- 1 root shadow 1784 2020-11-06 22:58 /etc/shadow
-rw-r----- 1 root shadow 1784 2020-10-29 18:12 /etc/shadow-
```
NB: Here, 'la' shows files ctime.

Of course, the `base-passwd` package is also installed.

Base on `/var/lib/dpkg/info/*passwd*ist\` : 
```
# la /var/lib/dpkg/info/*passwd.*
-rw-r--r-- 1 root root   1120 2020-05-27 09:05 /var/lib/dpkg/info/base-passwd.list
-rw-r--r-- 1 root root   1238 2019-02-10 20:10 /var/lib/dpkg/info/base-passwd.md5sums
-rwxr-xr-x 1 root root   2802 2019-02-10 20:10 /var/lib/dpkg/info/base-passwd.postinst
-rwxr-xr-x 1 root root    211 2019-02-10 20:10 /var/lib/dpkg/info/base-passwd.postrm
-rwxr-xr-x 1 root root   1516 2019-02-10 20:10 /var/lib/dpkg/info/base-passwd.preinst
-rw-r--r-- 1 root root 108623 2019-02-10 20:10 /var/lib/dpkg/info/base-passwd.templates
-rw-r--r-- 1 root root    134 2018-07-27 10:07 /var/lib/dpkg/info/passwd.conffiles
-rw-r--r-- 1 root root  12326 2020-05-27 09:05 /var/lib/dpkg/info/passwd.list
-rw-r--r-- 1 root root  20584 2018-07-27 10:07 /var/lib/dpkg/info/passwd.md5sums
-rwxr-xr-x 1 root root   1625 2018-07-27 10:07 /var/lib/dpkg/info/passwd.postinst
-rwxr-xr-x 1 root root   1044 2018-07-27 10:07 /var/lib/dpkg/info/passwd.preinst
```
it might not be a direct bug but an interaction with I don't know what.

XECDesign, 2020-11-09:
Is there any documentation stating that the shadow file is backed up each time passwd is run?

I'm having trouble seeing that anywhere in the source code or documentation. the shadow manpage just says the file is there and that some tools update it, but not much else.

JiffB, 2020-11-09:
@XECDesign 
Hmm, my bad, I dug my own doc and found an old note saying that /etc/shadow- is only updated when using vipw or update-passwd.
Sorry for the noise.

XECDesign, 2020-11-16:
No worries. I reached the same conclusion when digging through the source code of shadow tools, but was wondering if maybe I missed something.

closed 2020-11-16 by XECDesign