RPi-Distro/repo #210: Iptables v 1.8.2 contains a bug which prevents firewalld from being updated

adeuxdev2, 2020-11-10:
Buster: firewalld: after running any firewall rule changing command, and then reload the firewall, error messages are displayed.

This is caused by a bug in the iptables 1.8.2 package. See debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=931720

The solution is to upgrade the iptables package to version 1.8.3-2

adeuxdev2, 2020-11-11:
Just to add the latest version of iptables currently available in the repo is iptables 1.8.2-4.  This is a critical bug as it prevents the firewall from running.

XECDesign, 2022-01-19:
Looks like the issues was fixed a while back and Bullseye is on 1.8.7. Sorry this wasn't addressed when it was reported.

closed 2022-01-19 by XECDesign