RPi-Distro/repo #211: firmware-misc-nonfree package - USB wifi dongle not working on raspberry os [mt76x0u]

DaniloNC, 2020-11-18:
I had the same issue as reported on this forum post https://www.raspberrypi.org/forums/viewtopic.php?t=281071

Trying to use a mt76x0u based USB dongle gives the following error:
```
[ 6.721409] mt76x0u 1-1.2:1.0: ASIC revision: 76100002 MAC revision: 76502000
[ 6.724570] mt76x0u 1-1.2:1.0: mediatek/mt7610e.bin not found, switching to mediatek/mt7610u.bin
[ 6.724629] mt76x0u 1-1.2:1.0: Direct firmware load for mediatek/mt7610u.bin failed with error -2
[ 6.729700] mt76x0u: probe of 1-1.2:1.0 failed with error -2
```

The user MrEngman did find the reason for this bug:

```
Just thought I should explain the fix I used.

The mt7610 wifi driver is now included in RaspOS but it expects the wifi module firmware to be in a different directory to where it is currently installed. Why the wifi firmware directory has not been updated I don't know.

Anyway the commands in my previous post install the necessary wifi module firmware in the directory the wifi driver expects them to be in.

In the RaspOS the wifi module firmware is in directory /lib/firmware but it now needs to be in directory /lib/firmware/mediatek where the commands I used install it.

MrEngman
```

And had proposed a workaround:

```
I had similar problems getting the mt7610 wifi working on my Pi.

I managed to get it working using the following commands
Code: Select all

wget http://ftp.uk.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-misc-nonfree_20190114-2_all.deb
sudo dpkg -i firmware-misc-nonfree_20190114-2_all.deb
MrEngman
```

It would be good if this issue is fixed on the mainstream package.

kuanyili, 2021-06-22:
Source: firmware-nonfree
Maintainer: Serge Schneider <serge@raspberrypi.com>, @XECDesign

This bug is caused by missing firmware in `/lib/firmware/mediatek`, which should be fixed by upstream Debian in year 2019 in version `20190114-2`. Raspberry Pi OS is still based on old version `20190114-1`.

Raspberry Pi OS actually has both versions of `firmware-misc-nonfree` in built-in repos.

Run `apt show firmware-misc-nonfree --all-versions` and you'll see:

```
Version: 1:20190114-1+rpt11
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
```

```
Version: 20190114-2
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages
```

But unfortunately, the package from `archive.raspberrypi.org` has a version with an [epoch](https://www.debian.org/doc/debian-policy/ch-controlfields.html#version) of 1 so it is considered newer than the one from `raspbian.raspberrypi.org`.

### References

https://packages.debian.org/buster/firmware-misc-nonfree
https://metadata.ftp-master.debian.org/changelogs/non-free/f/firmware-nonfree/firmware-nonfree_20190114-2_changelog
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=919452

XECDesign, 2021-06-23:
Thanks, should be fixed by https://github.com/RPi-Distro/firmware-nonfree/commit/00de3194a96397c913786945ac0af1fd6fbec45b

kuanyili, 2021-06-23:
Many thanks for fixing it so quickly!

XECDesign, 2021-06-23:
No worries, only about a year too late.

closed 2021-06-23 by XECDesign