RPi-Distro/repo #213: install BRLTTY in standard distributions

scruss, 2020-11-26:
[BRLTTY](http://mielke.cc/brltty/) — a daemon which provides access to the Linux/Unix console in text mode for a blind person using a refreshable braille display — is not installed by default. I understand that it is a default part of the core Debian installation.

Please consider including it, full enabled, in the standard distributions. Without it, many blind users are shut out from using the Raspberry Pi.

For example: [Raspberry PI 400 and Braille display - Raspberry Pi Forums](https://www.raspberrypi.org/forums/viewtopic.php?f=102&t=292209)

XECDesign, 2020-11-26:
Makes sense. @spl237, since you've looked at accessibility in the past, any objections?

spl237, 2020-11-26:
As long as it's a) reasonably small so doesn't significantly increase the size of the image and b) not obviously visible to anyone who doesn't want to use it (no menu entry or obvious data files anywhere people will see them), then it's fine with me.

There is a question as to whether or not it interacts with the Orca screen reader, which also has Braille capability; we provide an easy way to install that (as of the next image, ctrl-alt-space installs it at any point) - I don't know anything about BRLTTY, so have no idea how it interacts with Orca's Braille features if at all, and I'd like to be assured it won't break Orca.

XECDesign, 2020-11-26:
I'll add it to the nightlies after the next image is out.

sthibaul, 2020-11-26:
@spl237 concerning orca vs BRLTTY, actually orca *uses* BRLTTY to be able to drive braille devices :) so no it won't break, on the contrary it needs it :)

spl237, 2020-11-26:
> @spl237 concerning orca vs BRLTTY, actually orca _uses_ BRLTTY to be able to drive braille devices :) so no it won't break, on the contrary it needs it :)

That's strange, because I did check dependencies to see if one depended on the other, and there is no dependency (in either direction) between the Orca and BRLTTY packages. I'd expect there to be one if Orca used BRLTTY.

sthibaul, 2020-11-26:
There is no dependency because a lot of people use Orca for speech output only, and thus don't install brltty.

spl237, 2020-11-26:
Ah - so are you saying that the Braille functionality in Orca won't work unless BRLTTY is also installed?

sthibaul, 2020-11-26:
Yes.

spl237, 2020-11-26:
Thank you - that has doubtless saved me several hours when I get around to trying to make Braille work in Orca and finding it doesn't...!

sthibaul, 2020-11-26:
btw, in case you didn't know: qemu supports virtual braille devices, there is some quickie documentation on

https://wiki.debian.org/DebianInstaller/Accessibility

spl237, 2020-11-26:
Thank you again - I'll have a play with it.

XECDesign, 2022-01-19:
Looks like we're still not shipping this. @spl237, did you have any luck with this?

Aside from pre-installing the package, is there anything that needs to be done? Seems like it installs and runs the service by default, but I don't have the hardware to test with.

sthibaul, 2022-01-19:
> I don't have the hardware to test with.

As mentioned above, qemu supports virtual braille devices

spl237, 2022-01-19:
If preinstalling it means it is running a service all the time, it probably doesn't make sense to have it preinstalled on all Pis. The best thing to do is probably for me to add it as a dependency to our Orca package; it would then be installed as part of the Ctrl-Alt-Space option prompted by the startup wizard.

XECDesign, 2022-01-19:
> As mentioned above, qemu supports virtual braille devices

Unless the situation has greatly improved in the recent years, running PiOS images under qemu is not the most trivial of tasks, so if somebody can say "yes that's all you need" or "you also need to make this other change", then that would be very helpful.

> If preinstalling it means it is running a service all the time, it probably doesn't make sense to have it preinstalled on all Pis.

Wouldn't help the 'lite' image users and I'd imagine that a console should inherently be much more accessible. Could disable it by default and add a "Hold <something> key to enable brltty" audio prompt like on the desktop.

spl237, 2022-01-19:
I've just tried adding it as a dependency for orca, and I'm afraid it breaks the screen reader completely - for some reason it stops orca running at startup. I have no idea why, and really don't have time to investigate at the moment.

If you think it is worth including in lite, then feel free to do so - unless we also offer orca on lite (I don't think we do...), in which case it is probably worth checking it doesn't break it on there as well!

XECDesign, 2022-01-19:
Alright, sounds like we'll need to shelve this one again. If I add it to lite, it will end up in the desktop image as well, aand if that breaks orca then it's not worth it right now.

spl237, 2022-01-19:
FWIW, we took advice from a leading accessibility consultancy a few years ago, and their advice was that it was far more important to have a screen reader than anything else - they didnt even mention compatibility with Braille hardware as a priority. In an ideal world we'd have both, but given it looks as if we can only have one of them, then orca is definitely the way to go for now.

spl237, 2022-01-27:
brltty has now been added as a dependency to the orca package, so it will be installed automatically along with orca. This can be trivially installed by pressing ctrl-alt-space at any time while the desktop is running.

XECDesign, 2022-02-16:
It probably doesn't make sense to re-implement the voice prompts and ctrl-alt-space keybind in the lite images, so I think we'll leave that as is.