RPi-Distro/repo #216: snmpd fail to install

swhall1, 2020-12-07:
I'm new to the raspian/apt world.  When trying to install snmpd on a pi3 the install fails because of missing mariadb files.  Is there a way to tell the pi to ignore the missing file and just upgrade to whats available?  

Specifically mariadb-common 10.0.3.25 doesn't even exist in archive but 10.0.3.27 does.  I tried install mariadb-common but that also references 10.0.3.25.  

Not sure what my options are at this time.  Does anyone have a recommendation?

thank you

XECDesign, 2020-12-08:
It seems to install fine here. What's the output of `apt policy snmpd`? Could you please provide some output showing the problem?

swhall1, 2020-12-08:
I got around the issue by force installing snmpd, mariad, libmariad and etc.  Burned a fresh copy of:
Linux respberrypi 5.4.79-v7+

Here is error I get:
pi@raspberrypi:~ $ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists... Done                                           
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 5d 16h 53min 9s). Updates for this repository will not be applied.
E: Release file for http://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 5d 1h 4min 18s). Updates for this repository will not be applied.
pi@raspberrypi:~ $ sudo apt install snmpd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libmariadb3 libsensors-config libsensors5 libsnmp-base libsnmp30 mariadb-common mysql-common
Suggested packages:
  lm-sensors snmp-mibs-downloader snmptrapd
The following NEW packages will be installed:
  libmariadb3 libsensors-config libsensors5 libsnmp-base libsnmp30 mariadb-common mysql-common snmpd
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 191 kB/4,121 kB of archives.
After this operation, 7,538 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf mariadb-common all 1:10.3.25-0+deb10u1
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf libmariadb3 armhf 1:10.3.25-0+deb10u1
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.25-0+deb10u1_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/m/mariadb-10.3/libmariadb3_10.3.25-0+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

swhall1, 2020-12-08:
And the policy output:
pi@raspberrypi:~ $ apt policy snmpd
snmpd:
  Installed: (none)
  Candidate: 5.7.3+dfsg-5+deb10u1
  Version table:
     5.7.3+dfsg-5+deb10u1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages

XECDesign, 2020-12-08:
`apt update` ran before the system time was updated, so it didn't work. Then `apt install` is trying to get files which no longer exist. Give it a few minutes to synchronise, then try `apt update` again. You shouldn't see those "InRelease is not valid yet" messages.