RPi-Distro/repo #218: audio problems in Firefox-ESR after PulseAudio is used

beta-tester, 2020-12-07:
hi,

after PulseAudio is used as top layer for audio devices, firefox-esr plays some videos without audio.
this wasn't an issue before the December 2020 RPi OS upgrade.

see here 

the update that was published today (2020-12-08) pulseaudio (12.2-4+deb10u1+rpi2) did not fix that issue.

how to reproduce that issue:
1. full-upgrade your RPi.
2. open firefox-esr
3. navigate to: https://www.youtube.com/watch?v=4Wrc4fHSCpw
4. play the video
(the video plays without audio)
5. open chromium
6. navigate to: https://www.youtube.com/watch?v=4Wrc4fHSCpw
7. play the video in chromium
(the video plays with audio)
8. open VLC media player
9. hit [Ctrl] + [N] and paste the following link: https://www.youtube.com/watch?v=4Wrc4fHSCpw
10. play the video in VLC media player
(the video plays with audio)

i tested on RPi4B (4GByte) and RPi3.
i tested full-upgrade (2020-12-04 and 2020-12-08) an previous RPi OS version (2020-08-20) and the current RPi OS version (2020-12-02)
all installations show up that issue.
the RPi's are connected to a TV by using HDMI cable.

spl237, 2020-12-08:
This seems to be a known / common issue with Firefox and PulseAudio; Firefox seems reluctant to use a running PulseAudio service. See https://bugzilla.mozilla.org/show_bug.cgi?id=1422073

You can fix it (in an not-entirely-satisfying way) by killing the existing PulseAudio service:

systemctl --user stop pulseaudio.service pulseaudio.socket

Then starting a new PulseAudio service in a terminal:

pulseaudio

And then running Firefox, which will then play audio via PulseAudio as expected.

As we are running an almost entirely vanilla PulseAudio instance - the only changes we have made are a couple of initial config changes which shouldn't have made any difference to this, and given the existing bug reports on similar-looking issues, this looks like something which Mozilla need to fix.

beta-tester, 2020-12-08:
i forgot to mention that not all video files are affected from this issue. most videos i played plays its audio well.
but some videos don't. and those which don't, don't do it reproducable.

e.g. https://www.youtube.com/watch?v=pTn6Ewhb27k
this video plays well with audio.

it is from the same author and channel and were published recently.

why is one playback with audio and the other not - constantly and so reproducable?
how can i see the audio format of the video of YouTube video?

the hint with killingand restarting PulseAudio broke the audio entirely of all applications - i had to reboot the RPi to have audio back. (at least to have the state as before with the issue of firefox-esr and no audio for some videos)

spl237, 2020-12-08:
To be honest, I'd suggest raising this as an issue against Firefox itself rather than here. We don't modify or customise Firefox - we just allow the standard package from Debian to be installed. As Chromium, VLC, etc all work properly with PulseAudio on Pi, finding out why Firefox behaves differently really won't be a priority for us, and we don't have anyone who knows the Firefox code and who could start to investigate without a very steep learning curve and a lot of time.

beta-tester, 2020-12-08:
unfortunately the mozilla guys are not very interested in maintaining the ARM version of firefox-esr as they told at an older issue (or at least not in raspberry pi)

ok, i will see how it goes this time...

beta-tester, 2020-12-16:
i opened an a bug report https://bugzilla.mozilla.org/show_bug.cgi?id=1682779

beta-tester, 2020-12-23:
> ... As Chromium, VLC, etc all work properly with PulseAudio on Pi ...

No VLC does NOT work properly with PulseAudio on RPi!
(tested on RPi4B and RPi3B with fresh image 2020-12-02-raspios-buster-armhf.zip)
see #217.
each audio or video file starts with two or more audio interruptions in the beginning.
this behavior is new since PulseAusio is used in Raspberry Pi OS.

beta-tester, 2021-03-29:
i opend an isuue report at https://bugs.launchpad.net/raspbian, because at Debian it is not an issue.
https://bugs.launchpad.net/raspbian/+bug/1921778

mrusli70, 2021-04-19:
I having the same problem with my Raspberry Pi 4B. And it too cannot run youtube on Firefox with raspbianOS version 5.10. Tested on Kali lInux works well running on youtube. 
Chromium and VLC works well. But not on Firefox with youtube.

spl237, 2021-04-19:
As above - please report as an issue with Mozilla against Firefox. We don't provide maintenance for Firefox, and audio does work with our supported browser, which suggests that this is a Firefox-specific issue. As per the link above, there are known problems with Firefox and pulseaudio.

closed 2021-04-19 by spl237

mrusli70, 2021-04-19:
Please be informed unable to submit Feedback to Firefox at this time as it indicates Please check back soon as they, mozilla paused submission, "Please check back soon".

beta-tester, 2021-04-20:
i it really an issue that mozilla can fix?
i opend an [issue report there](https://bugzilla.mozilla.org/show_bug.cgi?id=1682779) and did some logs they requested, but there was nothing that could explain the behavior.
and Firefox-ESR under other OS'es Debian, Ubuntu, Fedora, ... does work well in combination with PulseAudio.
only Raspberry Pi OS with PulseAudio seems to be affected from the issue at the moment. 
isn't the issue more somewhere in between Raspbian Pi OS dristro and Debian distro... that could be Raspbian.
there i opend an issue as well, but that report did not reached priority yet, maybe because nobory else reported and/or confirmed the issue there...
[Bug #1921778: 
audio problems for some videos in Firefox-ESR after PulseAudio is used  ](https://bugs.launchpad.net/raspbian/+bug/1921778)

beta-tester, 2021-05-22:
i guess YouTube recently changed their encoder settings. now nearly all videos have non-100% volume/normalized values. they are all at 95%/95%, that means Firefox-ESR on Raspberry Pi is absolute useless for YouTube videos

beta-tester, 2021-10-02:
there is a similar issue reported at https://github.com/webcompat/web-bugs/issues/88543

there the issue is with audio playback at the webpage https://sndtst.com/ where you normally can listen to in-game sound-tracks of some video-games.

on Raspberry Pi 10 Buster (2021-04-07) with chromium they play well, but with firefox-esr you will hear nothing at all.

beta-tester, 2021-11-09:
i tried the new RaspiOS Bullseye ("Normal" and Full) images on my RPi3B and RPi4B
and there it seems Firefox-ESR has its sound back again  :D

so it has been more a problem of how PulseAudio was integrated in Raspberry Pi OS Buster...
so no Firefox specific sound problem as it has been claimed again and again (at least for this specific issue).