RPi-Distro/repo #219: Raspberry Pi Desktop Taskbar flashes 6 times then disappears when using dead battery

79ad, 2020-12-17:
Compaq nx6110 laptop boots Raspberry Pi Desktop normally with good battery or with no battery. Booting with dead battery causes problem described. Battery needed as integral part of case.

OS appears functional but without GUI. As test: using "lxpanel" from command line shows 
errors but displays attenuated taskbar without menu.

Problem fixed in Jan/Feb 2020 after update but re-introduced a few months ago. 

Update/full-upgrade completed but did not resolve.

Battery parameters from BIOS:
BEN,57312 10/25/2005
C1,4000
C2,2271
C3,2116
ME,3
CC,1066
T,21
TV,11788
C,-914
DV,10800
MN,HP
CV1,0
CV2,3921
CV3,3936
CV4,3932
S,C0

spl237, 2020-12-17:
Something in the communication with a dead battery is crashing the interface to the battery monitor plugin, and without a device in a similar state to test, there's nothing I can realistically do to debug this.

The easiest fix is to remove the battery plugin from the taskbar, as it's pointless with a dead battery anyway. 

If the file ~/.config/lxpanel/LXDE-pi/panels/panel exists, edit it to remove the Plugin {} section which contains type ptbatt; it'll be near the end of the file. If your system doesn't have that file, then make the same edit to /etc/xdg/lxpanel/LXDE-pi/panels/panel, then reboot.

closed 2020-12-17 by spl237

79ad, 2020-12-18:
Hello Simon,

Thanks for your prompt response.

I had previously tried the edit on the first file (1) you suggested but not the second (2). However, I re-tried the edits since I had not repeated them since recent updates; here are the results:

Firstly I edited (1) changing the Plugin type to "ptbatt" from "batt" as it was different
Secondly I edited (1) removing the whole ptbatt/batt section
Thirdly I edited (2) removing the whole ptbatt section

Each time I rebooted but the results were the same.

I have attached some copies of the /home/pi/.cache/lxsession/LXDE-pi/run.log which may prove instructive; for example you can see the successive re-tries which give rise to the flashing in the "badbat" files.

I have included four files, two produced today to show the differences using a good battery and the bad one: 

run.log_goodbatafteredit
run.log_badbatafteredit

I have also included two files which I produced on 29 October. These were done using the "panel" files as installed.

run.log_goodbat
run.log_badbat

Regards,

Tony Dyke

spl237, 2020-12-18:
Unfortunately the attachments haven't made it to GitHub, but to be honest the output of lxsession probably won't reveal much.

Make sure you have the latest version of the package lxplug-ptbatt - it should be at version 0.14, as there was a fix pushed after the current image was created.

It might be worth trying this - when the panel has crashed, open a terminal window and do "lxpanel -p LXDE-pi"; you will presumably get a crash, but paste the terminal output before the crash here; it may reveal something.

But if the crash is happening even with ptbatt removed from both config files, then I have no idea what is happening; if that plugin doesn't even load, then there should be nothing communicating with the battery, so whether it is dead or not should make no difference to anything.

79ad, 2020-12-18:
Package lxplug-ptbatt is at version 0.14

pi@raspberrypi:~ $ lxpanel -p LXDE-pi
** Message: 12:39:38.082: x-terminal-emulator has very limited support, consider choose another terminal
** Message: 12:39:38.451: Connecting ...
** Message: 12:39:38.451: Status changed to down

** (lxpanel:2741): WARNING **: 12:39:38.452: lxpanel: can't load volumealsabt plugin

** (lxpanel:2741): WARNING **: 12:39:38.571: tray: another systray already running

** (lxpanel:2741): WARNING **: 12:39:38.572: lxpanel: can't load tray plugin
** Message: 12:39:38.580: Connecting ...
** Message: 12:39:38.580: Status changed to down

** (lxpanel:2741): WARNING **: 12:39:38.581: lxpanel: can't load volumealsabt plugin
Floating point exception
pi@raspberrypi:~ $ 

I am not sure why the attachments disappeared; I have sent them again, this time as individual files rather than a single 7zip file.

Perhaps I am not using GitHub appropriately and should be posting them directly rather than sending e-mail attachments; I created a logon specifically for this issue.

Tony

spl237, 2020-12-18:
From that, your configuration looks to be totally corrupted. You've got multiple copies of the same plugin loading; you're still loading the old volume plugin which is incompatible with the new PulseAudio environment, and in fact you're loading it twice! I'm not surprised things are crashing...

I'd suggest trying a clean copy of the latest image we released last week, because your current one looks a bit too broken to try to fix. Flash the latest image file to a USB device and boot from it with the battery disconnected, and choose Run with Persistence from the boot menu. When the desktop appears, right-click the taskbar, choose "Add/Remove Panel Items", and remove "Power & Battery" from the list. Shutdown, install the dead battery, boot up, see what happens.

79ad, 2020-12-21:
Simon,

Good Morning

I did what you suggested and it has been running properly for almost two days.

I couldn't find the latest image so  had to download to February release; it took hours to update!

I am not sure what caused the corruption; the laptop is just used as a Windows 7 replacement, though I do frequent updates; it just stopped working overnight.

Thanks for your analysis of the log files, it pushed me to do the fresh install. I am new to Linux and one in unsure which of the error messages are spurious and which arise from some, earlier, fundamental failure.

Thanks again for your help.

Regards,

Tony Dyke

spl237, 2020-12-21:
No problem - glad it is all working now.

Apologies for the mis-steer about the latest release - now I look, it's not been put up for download yet, but should be soon.