RPi-Distro/repo #220: udev rule for usb boot disk

frosch01, 2021-01-10:
I am using a PI4 with KDE Desktop. So not pure Raspberrypios but I am a real KDE fan and it works quite well with ~350MByte of memory in use on the idle desktop.

KDE uses udisks disk manager daemon as other desktops do, too. As an USB disk is registered, it will show up in the KDE device dialog for all users. So in the case of the boot and root partitions, these are offered to all the users as "pluggable" disks with options to mount, unmount etc. This is misleading and should be avoided.

I found that the device /dev/sda has the attribute "removable" set to 0. This makes sense and I assume the Linux kernel does this as it detects that the device is crucial to the system as it contains the root file system. I tested that USB pen drives have the removable flag set to 1, at least the ones I have. Perhaps there is an expert around that has clearer view on how the removable attribute is being built.   

By adding an udev rule to /etc/udev/rules.d/99-buitin_disks.rules I was able to mask the systemdisk from udisks. Perhaps it makes sense to add this rule to the RPI set of standard rules in general:

```
SUBSYSTEM=="block", ATTRS{removable}=="0", ENV{UDISKS_IGNORE}="1"
```

XECDesign, 2021-01-11:
Correct me if I am wrong, but I believe the attribute is inherent to the individual block device. For example, a flash stick would show up as removable even if it's your rootfs.

If you use the rule above, but actually run off an SD card, it would prevent the m.2 disk from showing up, which is not what you'd want in that case.

frosch01, 2021-01-24:
You are right, the removable flag is set by the USB->SATA adapter and set there to false. So my argumentation was wrong.

However, with the given rule the behavior is like on any other bus, e.g. on my PC are SATA devices not shown as removable media. So the question is if USB non removable devices should be shown in the plugable media list in general. I think it is more what users expect to not get such devices listed instead of having it on the list.

An improved rule could check if the UUID (via env ID_FS_UUID) or the device are listed inside the fstab using grep. This would require the root fs to be mounted already. This should be given as the udevd also requires to read it's configuration from /etc. So grep should also be available and rule could be improved like this.

What do you think?

XECDesign, 2021-01-25:
I don't personally have strong opinions but don't want to change things on a whim.

If I plugged that device into a pi to copy some files from it and then there was no eject button anywhere, I'd think that's wrong behaviour.

Being in fstab doesn't necessarily mean you never want to eject a device either. If it's rootfs, yeah sure.

On our desktop environment rootfs and boot don't show up. @spl237 is this something you specifically filter out or is udisks2 aware that they are system partitions?

spl237, 2021-01-25:
> On our desktop environment rootfs and boot don't show up. @spl237 is this something you specifically filter out or is udisks2 aware that they are system partitions?

As far as I am aware, I don't filter those - the GLib volume monitor knows they shouldn't be ejected when I query it. So I assume udisks2 knows they are system partitions.

XECDesign, 2021-01-25:
Thanks, Simon.

So then I'd say that if KDE displays the rootfs device in its ejector applet, it's doing something wrong.

frosch01, 2021-01-30:
Ok, so perhaps a case for KDE team. Thx guys,

closed 2021-01-30 by frosch01