RPi-Distro/repo #224: apt-get purge raspberrypi-artwork   leaves /usr/share/raspberrypi-artwork

dilyanpalauzov, 2021-01-27:
`apt-get purge raspberrypi-artwork` shall remove the file `/usr/share/raspberrypi-artwork/launch.png` and the directory `/usr/share/raspberrypi-artwork`, but does not.

I do not know where to file this.

XECDesign, 2021-01-28:
The package passes piuparts and there's nothing in the packaging that adds any files that wouldn't be managed by dpkg.

```$ sudo apt purge raspberrypi-artwork 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  raspberrypi-artwork*
0 upgraded, 0 newly installed, 1 to remove and 55 not upgraded.
After this operation, 110 kB disk space will be freed.
Do you want to continue? [Y/n] 
(Reading database ... 331066 files and directories currently installed.)
Removing raspberrypi-artwork (20150921) ...
$ ls -la /usr/share/raspberrypi-artwork
ls: cannot access '/usr/share/raspberrypi-artwork': No such file or directory
```

closed 2021-01-28 by XECDesign

dilyanpalauzov, 2021-01-28:
I install Rapsberry Pi OS with desktop, call `apt-get purge raspberrypi-artwork` and then the file /usr/share/raspberrypi-artwork/launch.png  does exist.

XECDesign, 2021-01-28:
Running `dpkg -S /usr/share/raspberrypi-artwork/launch.png` says that file is owned by raspberrypi-ui-mods