RPi-Distro/repo #226: Buster missing libasound.so?

shokuntsuki, 2021-02-03:
I have pyalsaaudio and libasound2-dev installed.  The code I am attempting to use has worked with previous installations of Raspbian.

As best I can tell, pyalsaaudio relies on /usr/lib/libasound.so being installed.  libasound.so is present in Debian's distribution but not in Buster.  Is there a reason for leaving it out of Buster?  (see https://larsimmisch.github.io/pyalsaaudio/pyalsaaudio.html#alsa-and-python)

XECDesign, 2021-02-04:
Everything looks okay from this end.

```
$ dpkg -S /usr/lib/arm-linux-gnueabihf/libasound.so
libasound2-dev:armhf: /usr/lib/arm-linux-gnueabihf/libasound.so
```

Which matches the paths in Debian:
```
/usr/lib/arm-linux-gnueabihf/libasound.so | libasound2-dev [armhf]
```
https://packages.debian.org/search?suite=buster&arch=any&mode=exactfilename&searchon=contents&keywords=libasound.so

shokuntsuki, 2021-02-12:
This did not turn out to be my problem, thank you!

closed 2021-02-12 by shokuntsuki