RPi-Distro/repo #227: Error in zfs-dkms with 5.10.11 kernel

bcarrella, 2021-02-06:
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.11+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.10.11+ /boot/kernel.img
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 5.10.11+ (armv7l)
Consult /var/lib/dkms/zfs/0.8.6/build/make.log for more information.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.11+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.11-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.10.11-v7+ /boot/kernel7.img
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 5.10.11-v7+ (armv7l)
Consult /var/lib/dkms/zfs/0.8.6/build/make.log for more information.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.11-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.11-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.10.11-v7l+ /boot/kernel7l.img
/etc/kernel/postinst.d/dkms:
Error! Bad return status for module build on kernel: 5.10.11-v7l+ (armv7l)
Consult /var/lib/dkms/zfs/0.8.6/build/make.log for more information.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.11-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.11-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.10.11-v8+ /boot/kernel8.img
/etc/kernel/postinst.d/dkms:
Error! echo
Your kernel headers for kernel 5.10.11-v8+ cannot be found at
/lib/modules/5.10.11-v8+/build or /lib/modules/5.10.11-v8+/source.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.11-v8+ /boot/kernel8.img

[make.log](https://github.com/RPi-Distro/repo/files/5935897/make.log)

bcarrella, 2021-02-09:
Same error was reported to Debian package maintainers:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=979158

XECDesign, 2021-02-10:
I don't think they're on 5.10, so they won't be able to help. We'd need to update the zfs package on our end.

bcarrella, 2021-02-11:
Thank you @XECDesign 
I can't access my external zfs usb drive at the moment. When the update will be available on apt repo?

XECDesign, 2021-02-11:
I've just added an updated version which should show up publicly within an hour.

bcarrella, 2021-02-11:
Problem is not resolved.
Maybe a kernel downgrade to 5.4.x can solve the issue?

[make.log](https://github.com/RPi-Distro/repo/files/5967411/make.log)

Meta:          1
Name:          zfs
Branch:        1.0
Version:       0.8.6
Release:      1~bpo10+1
Release-Tags:  relext
License:       CDDL
Author:        OpenZFS on Linux
**Linux-Maximum: 5.9**
Linux-Minimum: 2.6.32

XECDesign, 2021-02-11:
Not sure why you have 0.8.6. The updated version is 2.0.2.

bcarrella, 2021-02-11:
OK problem resolved purging and reinstalling zfs packages.
Now all zfs packages versions are 2.0.2
Thank you very much.

XECDesign, 2021-02-11:
Thanks for confirming everything is okay now. Not sure what went wrong, but I'll keep an eye on further reports of people having trouble upgrading.

closed 2021-02-11 by XECDesign