RPi-Distro/repo #228: wireguard-tools for RPi with kernel 5.10

kucharskim, 2021-02-06:
Hi,

Now that Raspberry Pi migrated to kernel 5.10, wireguard module is provided by default:

```
# uname -rv
5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021

# modinfo wireguard | wc -l
12
```

I was wondering would Raspberry Pi Engineering (not sure how to name the team developing Raspberry Pi OS) provide wireguard-tools packages in their own apt repositories, until Raspberry Pi migrates to Debian Bullseye as its base. I'm not even sure is development happening that Debian stable is the base for Raspberrry Pi OS.

```
# apt show wireguard-tools
N: Unable to locate package wireguard-tools
N: Unable to locate package wireguard-tools
E: No packages found
```

If above is not the case, what would be the least effort way today (today means kernel 5.10 on rpi) to install wireguard-tools (without compilation). I'm asking as I wouldn't like to manage release cycle of wireguard-tools myself, but leverage already existing apt repository with binary packages.

Regards,
 Mikolaj

Initially reported at [Raspberry Pi Forum](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=302834) but was pointed out to this GitHub repository.

XECDesign, 2021-02-15:
I've just added the wireguard packages from buster-backports. They should be live within an hour.

kucharskim, 2021-02-15:
Thanks, it's now available. It pulls `wireguard-dkms`, as dependency. I guess because it's part of buster, there is no easy way to disable or fake that dependency on RPi packages end? Could `raspberrypi-kernel` provide `wireguard-modules`? I'm not sure would that make sense, just asking.

```
# apt show wireguard-tools
Package: wireguard-tools
Version: 1.0.20200827-1~bpo10+1
Priority: optional
Section: net
Source: wireguard
Maintainer: Daniel Kahn Gillmor <dkg@fifthhorseman.net>
Installed-Size: 285 kB
Depends: libc6 (>= 2.8)
Recommends: nftables | iptables, wireguard-modules (>= 0.0.20171001) | wireguard-dkms (>= 0.0.20191219)
Suggests: openresolv | resolvconf
Homepage: https://www.wireguard.com
Download-Size: 76.9 kB
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
Description: fast, modern, secure kernel VPN tunnel (userland utilities)
...
```

XECDesign, 2021-02-15:
Since it's just a 'recommends' you should be able to remove it or avoid installing it with `--no-install-recommends` without any issues. But I suppose I could remove those dependencies altogether since it's enabled for all the kernels we ship.

kucharskim, 2021-02-15:
Sorry, I just didn't read carefully. Obviously it's recommends and not depends.

XECDesign, 2021-02-15:
Looks like Debian are 'providing' it in the kernel package:
https://salsa.debian.org/kernel-team/linux/-/blob/master/debian/templates/control.image.meta.in#L4

So I'll go for that approach.

kucharskim, 2021-02-15:
Great, thank you!

kucharskim, 2021-02-16:
Note to myself: `apt show raspberrypi-kernel`

XECDesign, 2021-03-03:
An update containing this change has just gone live.

closed 2021-03-03 by XECDesign

kucharskim, 2021-03-03:
Yup, all looks good. Thank you!

```
# uname -rv
5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021
```
```
# apt show raspberrypi-kernel
Package: raspberrypi-kernel
Version: 1.20210303-1
Priority: optional
Section: kernel
Source: raspberrypi-firmware
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 283 MB
Provides: linux-image, wireguard-modules (= 1.0.0)
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Homepage: https://github.com/raspberrypi/firmware
Download-Size: 78.9 MB
APT-Manual-Installed: no
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
```

Before:

```
# apt install -s wireguard-tools
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  build-essential cpp cpp-8 dkms dpkg-dev fakeroot g++ g++-8 gcc gcc-8
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libasan5 libc-dev-bin libc6-dev libcc1-0 libfakeroot libgcc-8-dev libisl19
  libmpc3 libmpfr6 libstdc++-8-dev libubsan1 linux-libc-dev
  raspberrypi-kernel-headers wireguard wireguard-dkms
Suggested packages:
  cpp-doc gcc-8-locales python3-apport menu debian-keyring gcc-8-doc
  libstdc++6-8-dbg gcc-multilib autoconf automake libtool flex bison gdb
  gcc-doc libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg libasan5-dbg
  liblsan0-dbg libtsan0-dbg libubsan1-dbg libmpx2-dbg libquadmath0-dbg
  glibc-doc libstdc++-8-doc openresolv | resolvconf
The following NEW packages will be installed:
  build-essential cpp cpp-8 dkms dpkg-dev fakeroot g++ g++-8 gcc gcc-8
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libasan5 libc-dev-bin libc6-dev libcc1-0 libfakeroot libgcc-8-dev libisl19
  libmpc3 libmpfr6 libstdc++-8-dev libubsan1 linux-libc-dev
  raspberrypi-kernel-headers wireguard wireguard-dkms wireguard-tools
0 upgraded, 29 newly installed, 0 to remove and 0 not upgraded.
Inst libc-dev-bin (2.28-10+rpi1 Raspbian:stable [armhf])
Inst linux-libc-dev (4.18.20-2+rpi1 Raspbian:stable [armhf])
Inst libc6-dev (2.28-10+rpi1 Raspbian:stable [armhf])
Inst libisl19 (0.20-2 Raspbian:stable [armhf])
Inst libmpfr6 (4.0.2-1 Raspbian:stable [armhf])
Inst libmpc3 (1.1.0-1 Raspbian:stable [armhf])
Inst cpp-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst cpp (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Inst libcc1-0 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst libasan5 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst libubsan1 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst libgcc-8-dev (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst gcc-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst gcc (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Inst libstdc++-8-dev (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst g++-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Inst g++ (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Inst dpkg-dev (1.19.7 Raspbian:stable [all])
Inst build-essential (12.6 Raspbian:stable [armhf])
Inst dkms (2.6.1-4 Raspbian:stable [all])
Inst libfakeroot (1.23-1 Raspbian:stable [armhf])
Inst fakeroot (1.23-1 Raspbian:stable [armhf])
Inst libalgorithm-diff-perl (1.19.03-2 Raspbian:stable [all])
Inst libalgorithm-diff-xs-perl (0.04-5+b1 Raspbian:stable [armhf])
Inst libalgorithm-merge-perl (0.08-3 Raspbian:stable [all])
Inst raspberrypi-kernel-headers (1.20210201-1 Raspberry Pi Foundation:testing [armhf])
Inst wireguard-dkms (1.0.20210124-1~bpo10+1 Raspberry Pi Foundation:testing [all])
Inst wireguard-tools (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [armhf])
Inst wireguard (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [all])
Conf libc-dev-bin (2.28-10+rpi1 Raspbian:stable [armhf])
Conf linux-libc-dev (4.18.20-2+rpi1 Raspbian:stable [armhf])
Conf libc6-dev (2.28-10+rpi1 Raspbian:stable [armhf])
Conf libisl19 (0.20-2 Raspbian:stable [armhf])
Conf libmpfr6 (4.0.2-1 Raspbian:stable [armhf])
Conf libmpc3 (1.1.0-1 Raspbian:stable [armhf])
Conf cpp-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf cpp (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Conf libcc1-0 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf libasan5 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf libubsan1 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf libgcc-8-dev (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf gcc-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf gcc (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Conf libstdc++-8-dev (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf g++-8 (8.3.0-6+rpi1 Raspbian:stable [armhf])
Conf g++ (4:8.3.0-1+rpi2 Raspbian:stable [armhf])
Conf dpkg-dev (1.19.7 Raspbian:stable [all])
Conf build-essential (12.6 Raspbian:stable [armhf])
Conf dkms (2.6.1-4 Raspbian:stable [all])
Conf libfakeroot (1.23-1 Raspbian:stable [armhf])
Conf fakeroot (1.23-1 Raspbian:stable [armhf])
Conf libalgorithm-diff-perl (1.19.03-2 Raspbian:stable [all])
Conf libalgorithm-diff-xs-perl (0.04-5+b1 Raspbian:stable [armhf])
Conf libalgorithm-merge-perl (0.08-3 Raspbian:stable [all])
Conf raspberrypi-kernel-headers (1.20210201-1 Raspberry Pi Foundation:testing [armhf])
Conf wireguard-dkms (1.0.20210124-1~bpo10+1 Raspberry Pi Foundation:testing [all])
Conf wireguard-tools (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [armhf])
Conf wireguard (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [all])
```

After:

```
# apt install -s wireguard-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  openresolv | resolvconf
The following NEW packages will be installed:
  wireguard-tools
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Inst wireguard-tools (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [armhf])
Conf wireguard-tools (1.0.20200827-1~bpo10+1 Raspberry Pi Foundation:testing [armhf])
```