RPi-Distro/repo #229: ip6_gre module is not included in distributed Raspberry Pi OS Kernel Modules

owendelong, 2021-02-07:
The ip_gre module is included in /lib/modules/.../ipv4

However, the ip6_gre module is missing from /lib/modules/.../ipv6

Please correct this or at the very least create an ipv6 kernel add-ons package that allows this and any other missing iPv6 kernel modules and support utilities to be easily pulled in.

owendelong, 2021-02-07:
Looking closer, it appears all of the following modules are missing from .../ipv6:

esp6_offload.ko
fou6.ko
ila/ila.ko
ip6_gre.ko
ip6_vti.ko

And from .../ipv6/netfilter:
ip6table_security.ko
ip6t_srh.ko
ip6t_SYNPROXY.ko

XECDesign, 2021-02-08:
I believe this is a kernel defconfig issue? If so, it should be reported here - https://github.com/raspberrypi/linux/

owendelong, 2021-02-13:
Well... It could be solved by changing the kernel defconfig, or, it could be solved by making the additional modules available as an add-on package. I'd prefer it be solved by modifying the default config, frankly, but I put it here because it's kind of a bistro-decision about where you want to fix it first, then it should go to the appropriate place based on that decision.

owendelong, 2021-02-13:
Just in case, have opened https://github.com/raspberrypi/linux/issues/4144 based on your feedback.

XECDesign, 2022-01-19:
Looks like this is included now.

closed 2022-01-19 by XECDesign