RPi-Distro/repo #232: Fix WPA2 Enterprise Wifi Module

Picajoluna, 2021-02-15:
Hello,

I'm a college student who owns a Raspberry Pi Zero W. I've struggled to get my RPi connected to my school's WPA2 Enterprise network, until I very recently came across this [fix on a forum post](https://www.raspberrypi.org/forums/viewtopic.php?p=1601893&sid=9154e8b5618a5ddb5054191a15ea3af4#p1601893), which involves a very simple swap of `nl80211,wext` for `wext,nl80211` in the `/lib/dhcpcd/dhcpcd-hooks/10-wpa_supplicant` system file.

It took me forever to find this fix, until then I was using a script that I found elsewhere to downgrade the wpa_supplicant package to an older version, which is not ideal.  I would love to issue a pull request to suggest the change and prevent other folks this needless headache, but I can't find the GitHub repository which contains that file / the repository in which I can issue the PR. Does anyone know where I could start that conversation, if this is the wrong place for it?
Thanks.

XECDesign, 2021-02-15:
The change is made directly to the dhcpcd package rather than the net or sys mods packages, so I'll move the issue here.

I think the thing to do is to revert this change for the bullseye release when some breakage is expected, then see what happens.

XECDesign, 2022-01-12:
The current package (which is closer to Debian's) doesn't have the driver defined, which should resolve the issue.

closed 2022-01-26 by no-response[bot]