RPi-Distro/repo #236: Error! The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which...

pheiduck, 2021-03-04:
Background:

Use wireguard installed with this script https://github.com/Nyr/wireguard-install

Wireguard is working anyway

Get this from todays Kernel Update updated via apt / stable buster repo > 1.20210303-1
```
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7l+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v8+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
```

pheiduck, 2021-03-04:
https://github.com/raspberrypi/firmware/issues/1545

XECDesign, 2021-03-04:
That script doesn't seem to support Raspberry Pi OS. AFAIK, Wireguard is already provided as a kernel module and you can just run `sudo apt install wireguard-tools` to get the tools

XECDesign, 2021-03-04:
Taking a closer look, it seems like that message comes from the `wireguard-dkms` package, which you should be able to remove.

pheiduck, 2021-03-04:
hm, sure that wireguard won't work if I do this

XECDesign, 2021-03-04:
> hm, sure that wireguard won't work if I do this?

Why do you believe that?

pheiduck, 2021-03-10:
Update: uninstalled wireguard-dkms dkms
Works anyway

closed 2021-03-10 by pheiduck