RPi-Distro/repo #239: missing security update for firefox-esr 78.9

beta-tester, 2021-03-29:
hi,
i am using firefox-esr on my RPi 4.
last week (2021-03-23) a security update for firefox-esr was released by Mozilla.
Debian released the update for armhf one day later (2021-03-24) at their repository.
(according the date at [Debian pool](http://deb.debian.org/debian/pool/main/f/firefox-esr/?C=M;O=D))
for my RPi 4 on raspberry repository i still can not update/upgrade to that security update.

XECDesign, 2021-03-29:
It does seem to be out of date, but it's in in the Raspbian repo, which we have no control over.

Since that package has some pi-specific patch added to it, it may be that it didn't build automatically and the Raspbian maintainer needs to manually update his changes for the new version. Since it has only been a few working days, I wouldn't worry about it just yet, but if you'd like you could try reporting the issue here: https://bugs.launchpad.net/raspbian

beta-tester, 2021-03-29:
i am a little bit confused...
as @spl237 told me at issue #218 raspberry pi os distro is not responsable for that issue because firefox-esr is a unmodified version of the debian version
_"We don't modify or customise Firefox - we just allow the standard package from Debian to be installed"_

spl237, 2021-03-29:
The Raspbian maintainer sometimes has to make small changes to packages so they will build for armhf. He makes no functional changes to the code. He will need to make whatever changes he has to previous versions to enable the new version to build, at which point the package will appear in the Raspbian repository.

XECDesign, 2021-03-29:
Right, there's some conflation between Raspbian and Raspberry Pi OS.

We, people who run archive.raspberrypi.org and are responsible for Raspberry Pi OS, do not modify Firefox in any way. Raspberry Pi OS is built on top of raspbian (mirrordirector.raspbian.org), but we have no control over packages there. When we want to fix something like this, we normally add a package to our repo to override the one that comes from Raspbian. Since we already support Chromium, we don't have the resources to take on another web browser.

The whole setup makes reporting bugs a bit tricky, but in general you need to go up the chain to where the bug is introduced. If it's caused by a patch we've added, that's 100% on us. If it comes from a change added by Raspbian, that's on them. If the bug also exists in Debian, then the bug should be reported to them.

beta-tester, 2021-03-29:
ok, thank you... now i reported #218 at https://bugs.launchpad.net/raspbian, because at Debian it is not an issue.

closed 2021-03-29 by beta-tester

beta-tester, 2021-03-31:
update `firefox-esr armhf 78.9.0esr-1~deb10u1+rpi1` just arrived.
so it takes about one week.
next time i will wait a bit longer befor i cry. sorry.