RPi-Distro/repo #248: R Pi Configuration GUI tool, 'boot to pi' is incorrect name after usermod -l

phillHgl, 2021-06-15:
Raspbery Pi Buster, latest updates, Pi 4b 8GB with OS loaded by recent Canakit purchase.  (very impressed!)  However based on security recommendations of several sights I:
a) Used Raspbaerry Pi Configuration to disable 'boot to pi'
b) used useradd and groupadd to create a user and a primary group with a 'newname'
c) used usermod to 1) copy all files in /home/pi to a new home dir path using newname
d) used usermod to rename pi to newname, and associate the new home dir and new primary group to the renamed user (which still has the original 'pi' user ID).
After testing these changes successfully, in Raspberry Pi Configuration tool observed that it still says 'boot to pi' even though the 'pi' user does not exist.  I went ahead and selected this option and it does boot to my renamed user.  So my report is that the GUI is misleading by not displaying the correct account name.  However I also recognize that this is mostly an aesthetics issue as the function is as desired, and I recognize that it may be difficult to design a dialog that could accommodate an account name of MAXLENGTH

XECDesign, 2021-06-15:
I'm not seeing a "boot to pi" string anywhere in raspi-config.

Everything references "$USER", which is determined by:

```
USER=${SUDO_USER:-$(who -m | awk '{ print $1 }')}
```

XECDesign, 2022-01-19:
@spl237 Actually, maybe I closed this one too soon. Does rc-gui do the right thing? I may have seen it refer to 'pi' when the user was renamed.

Although some upcoming changes may make this irrelevant anyway.

closed 2022-01-19 by XECDesign

spl237, 2022-01-19:
The text used was changed many years ago to say "current user" rather than "pi" in the UK English translation file. Translations for other countries - including the USA - will need to be updated accordingly, but we only manage the UK English file; we rely on user submissions for other countries.