RPi-Distro/repo #249: RPiOS Desktop is missing ImageMagick's display, so PIL.Image.show() Python examples won't work

scruss, 2021-06-24:
**Describe the bug**
System-provided Python Imaging Library assumes there's an image viewer (one of ImgaeMagick's display, Gnome's eog, or xv) installed, but it's not present. Some RPF publications (the Camera Guide, amongst others)  and many forum posts hit this problem.

A typical problem thread: [Display jpg image using Image Viewer python 2.7 - Raspberry Pi Forums](https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=314427). There are reports of this going back to 2014, possibly further.

**To reproduce**
Run this code in Python or Python 3, substituting a real image path:

    from PIL import Image
    img = Image.open('butterfly1.jpg')
    img.show()

**Expected behaviour**
Window showing the image opens.

**Actual behaviour**
Nothing: no window, no error messages. Running the command through strace shows that display, eof and xv are being searched for in the PATH, but not found.

**Remedy**

Provide the imagemagick binary package (`sudo apt install  imagemagick`) with Raspberry Pi OS with Desktop. 

**System**

Raspbian GNU/Linux 10 (buster) on Raspberry Pi 4B (but seems universal to Desktop users).

XECDesign, 2021-06-24:
Might make more sense to patch PIL to also try using the default image viewer we use, or is there a reason imagemagick is preferable?

scruss, 2021-06-24:
I only suggested ImageMagick's `display` because there's already some of ImageMagick installed with the default Desktop image: at least *libmagickcore*, possibly others. While installing `display` does bring in a few more dependencies, it's:

* likely smaller than bringing in all the Gnome support that `eog` needs. (NB: I typo'd this as `eof` in the main issue text, will fix)
* free and maintained some time this century, unlike xv.

You *could* patch PIL, but you'd have to make sure that it's patched all the way back to pypi for both Python 2 and Python 3 to ensure that it'll work no matter how people come to it. Despite the warnings, while Python 2 and all the ancient tutorials are still available, people will still use it.

Rather than patching it to the LXDE/Raspberry Pi OS default of `gpicview`, PIL should really be using `xdg-open`. This is the freedesktop.org "open this thing with the default viewer" command. Under Raspberry Pi OS, it uses gpicview. On Ubuntu, it uses eog. I'll see if I can make any noise upstream on PIL to see if they'll make this change. I can't poke Debian, as I've no installed system, but I'll see what Ubuntu-bug can do for me.

Thanks for the prompt response!

XECDesign, 2021-06-24:
I wouldn't be concerned about patching it to pypi, since we only support apt packages.

> After this operation, 14.5 MB of additional disk space will be used.

Imagemagick wouldn't take up too much extra space, so maybe:

I've just checked and `display` is provided through the 'update-alternatives' mechanism. So maybe the thing to do would be to make gpicview register itself as a provider of `display` and that will solve the issue without eating up extra space, adopting an extra package to maintain or worrying about pypi.

XECDesign, 2021-07-08:
Should be fixed

closed 2021-07-08 by XECDesign