RPi-Distro/repo #250: The debian/copyright file is missing in rpi-update

zhan-kunbus, 2021-06-29:
According to the Debian Policy Manual, the Copyright info is needed, which can be put in file debian/copyright.

However, should a repository be created in RPi-Distro and put the debian files in it ?

zhan-kunbus, 2021-07-05:
Can the firmware, kernel & modules be well managed/updated by the apt package management **without** the rpi-update ?
It looks to me that, most of the files are contained in the package raspberrypi-bootloader, by which most files updated by rpi-update are managed.

XECDesign, 2021-07-05:
Yes, rpi-update came along before there was an apt package. It has the advantage of being able to get the latest version before it goes to the `stable` branch and lets you switch between different versions, but in general it's not necessary for the average user.

I will add the copyright, but this isn't a package that we're trying to get into Debian, so adhering to Debian's policies is not particularly important.

XECDesign, 2021-07-05:
Should be sorted

closed 2021-07-05 by XECDesign