RPi-Distro/repo #253: Add raspberrypi-archive-keyring to bullseye suite

MichaIng, 2021-08-07:
The buster suite contains the nice `raspberrypi-archive-keyring` package to install and potentially update the keyring for the Raspberry Pi repository. The bullseye suite's `Packages` files currently do not contain it yet, while the one existing package from the repository pool works fine.

I know this package is not even used by pi-gen nor shipped with Raspberry Pi OS, but actually I would vote for using it to store keyrings in dedicated `/etc/apt/trusted.gpg.d/*.gpg` files, instead of storing them within the deprecated `/etc/apt/trusted.gpg`. At least for manual installs, it would be great to have this package in the bullseye suite.

beta-tester, 2021-12-02:
is this `raspberrypi-archive-keyring` package related to `gnome-keyring` package i need to install for remembering passphrase for access to encrypted partitions?
see #266

MichaIng, 2021-12-02:
No these are not related at all. `raspberrypi-archive-keyring` provides the public key for the `archive.raspberrypi.org` APT repository, i.e. `/etc/apt/trusted.gpg.d/raspberrypi.gpg`. It is not required on Raspberry Pi OS images, since those have this key added to `/etc/apt/trusted.gpg` already (which is btw deprecated, hence this shouldn't be done anymore). Without this key, you will face errors when running `apt update`.

XECDesign, 2022-01-13:
I never quite understood the point of it. How's the package used? If need the keys contained within the package to install the package...?

MichaIng, 2022-01-13:
Indeed a chicken & egg issue, though consistent with Debian and Raspbian repositories, especially also when it would be used for Raspberry Pi OS images in the first place. It feels cleaner to me when files are part of a package (might sound funny, looking at DietPi, we aim to go there 😅), it gives you the chance to transition to a new key, when wanted or required, by shipping two keys for transition with the package etc.

XECDesign, 2022-01-13:
For debian and raspbian it makes sense because those packages should be installed as a part of the bootstrap process. Not so much for additional repos.

I'll add it anyway, just wanted to make sure there wasn't some utility for it that I wasn't aware of.

XECDesign, 2022-01-13:
Would this work for your use case? http://archive.raspberrypi.org/debian/pool/untested/r/raspberrypi-archive-keyring/raspberrypi-archive-keyring_2021.1.1+rpt1_all.deb

MichaIng, 2022-01-13:
What is the purpose of the extra differentiation/copy of `raspberrypi-archive-raspberrypi.gpg` vs `raspberrypi-archive-keyring.gpg`?

Just for comparison, the package from the Buster suite contains:
```
# dpkg -L raspberrypi-archive-keyring
/.
/etc
/etc/apt
/etc/apt/trusted.gpg.d
/etc/apt/trusted.gpg.d/raspberrypi.gpg
/usr
/usr/share
/usr/share/doc
/usr/share/doc/raspberrypi-archive-keyring
/usr/share/doc/raspberrypi-archive-keyring/README
/usr/share/doc/raspberrypi-archive-keyring/changelog.gz
/usr/share/doc/raspberrypi-archive-keyring/copyright
/usr/share/keyrings
/usr/share/keyrings/raspberrypi-archive-keyring.gpg
```
I personally would prefer it to use `/etc/apt/trusted.gpg.d/raspberrypi-archive-keyring.gpg` as a symlink to `/usr/share/keyrings/raspberrypi-archive-keyring.gpg`, to avoid copies of files while admins can still alter of remove `/etc/apt/trusted.gpg.d/raspberrypi-archive-keyring.gpg` as it's defined as config file and hence not automatically updated by APT when modified or removed.

But yes, it would work pretty fine 👍.

XECDesign, 2022-01-13:
No idea, it's just the modified Debian package.

https://packages.debian.org/bookworm/all/debian-archive-keyring/filelist

I'm guessing that keyring.gpg contains all the keys and the other ones just contain the individual keys. The postrm file looks like there should be a symlink, but I don't see it trying to create one anywhere.

XECDesign, 2022-01-13:
> Install separate keyrings for each release in /usr/share/keyrings
    (both in the regular deb and the udeb).  This is to enable
    initial testing of narrowing the set of trusted keys per machine.

Not really relevant for Raspberry Pi OS, but no point diverging from Debian's direction either.

XECDesign, 2022-01-13:
Heads up that I've changed the name to `raspberrypi-archive-stable.gpg` so it's a bit less ugly. Should be public soon.

closed 2022-01-14 by MichaIng