RPi-Distro/repo #258: Update repository suite names

MichaIng, 2021-09-01:
While Debian Bullseye has been released, and also Raspbian shows it correctly as "stable" and Buster as "oldstable"
- http://raspbian.raspberrypi.org/raspbian/dists/buster/Release
- http://raspbian.raspberrypi.org/raspbian/dists/bullseye/Release

the RPi repository still shows Buster as "testing" and Bullseye as "oldstable":
- http://archive.raspberrypi.org/debian/dists/buster/Release
- http://archive.raspberrypi.org/debian/dists/bullseye/Release

It is more a formality and has no practical impact, but I just got a question about how to remove the unwanted testing repositories from the system, so at least this caused confusion in one case now 😄.

I think it should be trivial to have this updated and aligned with Raspbian, respectively the actual Debian releases.

XECDesign, 2021-09-01:
We'll update that once we release bullseye.

MichaIng, 2021-09-01:
Ah yes, makes sense to wait for an actual Raspberry Pi OS Bullseye release. It just lacks behind one cycle already, hence the report.

> that people don't get the error about the suite changing when they use apt-get or the GUI tools

There is a prompt to accept the suite change, but no error. But non-interactive calls without STDIN will fail, unless `--allow-releaseinfo-change-suite` or `--allow-releaseinfo-change` is passed.

XECDesign, 2021-09-01:
IIRC, when running apt there's a prompt, when running apt-get it's an error. I know it's an easy fix, but it throws a lot of people off.

MichaIng, 2021-09-01:
To be sure, I just tested both on a fresh Raspberry Pi OS, and to my surprise, both, `apt update` and `apt-get update` now show a notice only and don't require any interactive confirmation. This was definitely different with earlier APT versions, when Buster was released:
```
pi@raspberrypi:~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [380 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Fetched 13.5 MB in 18s (764 kB/s)
Reading package lists... Done
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
```
```
pi@raspberrypi:~ $ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [380 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Fetched 13.5 MB in 19s (713 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
45 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
```
Luckily, the Raspberry Pi OS image comes with present APT list files, which I'd normally call a waste of space and traffic, but I couldn't find any other system of image with list files and without having the suite chance done already 😄.

A reasonable change as, when using codenames in `sources.list`, a suite change doesn't imply any risk, compared to a codename change when using suites in `sources.list`, where a distro upgrade would be applied.

XECDesign, 2021-09-02:
IIRC, some commands use those lists for tab completion of package names. When I took them out a while back, it upset some people that they had to run `apt update` first.

> To be sure, I just tested both on a fresh Raspberry Pi OS, and to my surprise, both, `apt update` and `apt-get update` now show a notice only and don't require any interactive confirmation.

Oh good. Then I think it's just GUI tools like "Recommended Software" that have the issue. AIUI, they use `packagekit` as a backend which doesn't handle this.

MichaIng, 2021-09-02:
> When I took them out a while back, it upset some people that they had to run `apt update` first.

So they prefer to tab through an outdated package list instead of doing the no. 1 first command admins run on a fresh APT based system 🤔. But indeed `apt list` and `apt install` indeed use the package cache, and if not present probably the raw lists as well.

> Then I think it's just GUI tools like "Recommended Software"

Possibly, with some luck or a bug tracker report they may adopt `apt`'s behaviour.

XECDesign, 2021-09-02:
> So they prefer to tab through an outdated package list instead of doing the no. 1 first command admins run on a fresh APT based system.

Yeah... It's either go with what people are actually used to doing, or have the same forum threads and emails popping up about how we've broken everything.

MichaIng, 2021-11-09:
Suite names have been updated, now matching those of Raspbian and Debian repositories 👍. I'll mark this hence as closed.

closed 2021-11-09 by MichaIng