RPi-Distro/repo #260: Usb mount issue

symbios24, 2021-09-20:
Hello
I currently have an issue with latest raspbian in a Raspberry pi 4
In first boot it auto mounts usb flash drives in the second or third boot stops auto mount any incerted flash drives if i incert a flash drive and press open in the dialog that says open with file manager then i get an error that the directory /media/pi/disk/ is not valid

XECDesign, 2021-09-20:
@spl237

spl237, 2021-09-29:
I can't repeat this - repeatedly rebooting a Pi with an USB flash drive inserted mounts the drive every time. There's no difference from the point of view of the software and hardware between successive reboots, so there is no reason why there should be different remount behaviour.

It may well be that your USB drive is faulty - can you repeat the issue with a different drive?

closed 2022-01-19 by XECDesign