RPi-Distro/repo #263: openjdk-8-jre/jdk broken on armv*l

theofficialgman, 2021-09-23:
using the openjdk-8-jre package installed from the raspbian repo on armhf, running a simple java -version fails

the openjdk-11 package works just fine

```
/usr/lib/jvm/java-8-openjdk-armhf/bin/java -version

#

# A fatal error has been detected by the Java Runtime Environment:

#

#  SIGBUS (0x7) at pc=0xe63bc828, pid=29085, tid=0xe608a460

#

# JRE version:  (8.0_212-b01) (build )

# Java VM: OpenJDK Client VM (25.212-b01 mixed mode linux-aarch32 )

# Problematic frame:

# V  [libjvm.so+0x15e828]

#

# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again

#

# An error report file with more information is saved as:

# /home/garrett/hs_err_pid29085.log

#

# If you would like to submit a bug report, please visit:

#   http://bugreport.java.com/bugreport/crash.jsp

#

Aborted (core dumped)

```

theofficialgman, 2021-09-23:
```
#

# A fatal error has been detected by the Java Runtime Environment:

#

#  SIGBUS (0x7) at pc=0xe63bc828, pid=29085, tid=0xe608a460

#

# JRE version:  (8.0_212-b01) (build )

# Java VM: OpenJDK Client VM (25.212-b01 mixed mode linux-aarch32 )

# Problematic frame:

# V  [libjvm.so+0x15e828]

#

# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again

#

# If you would like to submit a bug report, please visit:

#   http://bugreport.java.com/bugreport/crash.jsp

#

---------------  T H R E A D  ---------------

Current thread (0xe5f07400):  JavaThread "Unknown thread" [_thread_in_vm, id=29086, stack(0xe603b000,0xe608b000)]

siginfo: si_signo: 7 (SIGBUS), si_code: 1 (BUS_ADRALN), si_addr: 0xe5f4fe0d

Registers:

R0=0x0000000a

R1=0xd3958170

R2=0xe5f4fe0d

R3=0xd3958150

R4=0x00000009

R5=0xe5f07400

R6=0xe6089820

R7=0x000000e5

R8=0x0000000a

R9=0xe686c000

R10=0xe686c000

R11=0xe6089454

R12=0xe5f4fe15

R13=0xe6089350

R14=0xe5f4fe0c

R15=0xe63bc828
[... 547 lines cut ...]
BogoMIPS	: 38.40

Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32

CPU implementer	: 0x41

CPU architecture: 8

CPU variant	: 0x1

CPU part	: 0xd07

CPU revision	: 1

Memory: 4k page, physical 4079184k(497564k free), swap 1048572k(934216k free)

vm_info: OpenJDK Client VM (25.212-b01) for linux-aarch32 JRE (1.8.0_212-8u212-b01-1+rpi1-b01), built on Mar 29 2019 13:54:02 by "root" with gcc 8.3.0

time: Thu Sep 23 03:37:36 2021

timezone: UTC

elapsed time: 0 seconds (0d 0h 0m 0s)

```

XECDesign, 2021-09-23:
You seem to be running this with an aarch64 kernel? IIRC, Java's cpu detection is pretty bad and it gets confused when you do that. Does it work if you run a 32bit kernel?

theofficialgman, 2021-09-23:
Unfortunately the system I am testing on I can’t easily run a 32bit kernel. I can say that this might have been fixed in upstream because I have used this binary package of java 8 on someones github that does work and seems to have a newer build version. https://github.com/gl91306/lunar/tree/master/jre

XECDesign, 2021-09-23:
If you're aware of a patch that fixes the issue, I could pull it in, but as is, we don't currently do our own builds, nor does the package come from our repo. You could try reproducing the issue running Debian, then reporting it there.

As is, this is a wontfix.

theofficialgman, 2021-09-23:
@XECDesign could you point out to me where the openjdk-8 builds are coming from on the raspbian repo? they aren't in the debian stable repos (debian stable doesn't have openjdk 8 on bullseye and buster) and I can't figure out exactly where the raspbian build comes from

to me it does seem that this is a custom build. this is the build number from raspbian 8u212-b01-1+rpi1

XECDesign, 2021-09-23:
They come from raspbian. but I believe they add all packages from the previous release into new ones. So the original source would've been debian + some patch from raspbian (indicated by the +rpi1 in the version number) and then it was left unmaintained.

XECDesign, 2021-09-23:
Given that it's not even in Debian, we'll definitely not be patching this even if there's a fix.

closed 2021-09-23 by XECDesign

theofficialgman, 2021-09-23:
If its unmaintained, broken, and old don’t you think it should be removed from the repo?

XECDesign, 2021-09-23:
Absolutely, but it's not in our repo.

https://bugs.launchpad.net/raspbian

theofficialgman, 2021-09-23:
what do you mean its not in your repo? its right here: http://archive.raspbian.org/raspbian/pool/main/o/openjdk-8/
right in the archive.raspbian.org repo and its included in the bullseye and buster and probably older versions as well (haven't checked all the release files)

XECDesign, 2021-09-23:
You're confusing Raspberry Pi OS and Raspbian. Raspbian is a community project that aims to rebuild Debian for armv6. Raspberry Pi OS, at least for armhf, is built on top of Raspbian, but they're completly separate projects.

This issue tracker is for Raspberry Pi OS. The repo for it is archive.raspberrypi.org, not archive.raspbian.org.

theofficialgman, 2021-09-23:
oh my bad... I got this issue location from this page on raspbian https://www.raspbian.org/RaspbianBugs

I should file it at the launchpad. my bad and thanks for the help that you have given me

XECDesign, 2021-09-23:
Best of luck