RPi-Distro/repo #264: Changes in vchiq driver in mainline require modified udev rules.

OjaswinM, 2021-09-23:
Hello,

Recently, there was change in the vchiq_arm staging driver in the mainline linux kernel. The patch is present here: https://lore.kernel.org/linux-staging/20210907115045.2206083-1-gregkh@linuxfoundation.org/#r

The above link can be followed for more details, but in short, the vchiq cdev now uses the miscdevice interface to register itself. This makes the udev rule for the vchiq cdev to break resulting in the following behavior:

```
* Permissions before the patch *
$ ls -l /dev/vchiq 
crw-rw---- 1 root video 235, 0 May  7 17:33 vchiq

* Permissions after the patch *
$ ls -l /dev/vchiq 
crw------- 1 root root 10, 125 May  7 17:30 vchiq
```
In the Pi OS distro I'm testing with, the rule in question is defined at `/lib/udev/rules.d/10-local-rpi.rules` as follows:

```
SUBSYSTEM=="vchiq", GROUP="video", MODE="0660" 
```
I was able to fix this by modifying the rule as follows:

```
KERNEL=="vchiq", GROUP="video", MODE="0660"
```
I tested this with and without this patch and it works as expected. 

So, I was looking to send a PR/patch to the repo which holds these rules for Pi OS, but unfortunately, I can't seem to find it. I tried to check the udev rules here [1] but can't find the vchiq specific file. I also checked the raspberry/userspace repo and found a (commented out) reference to possibly a vchiq related udev file [2], but again no luck finding the actual file.

Can someone please point out where the file exists and if its possible to send a PR for it. Also, please do let me know if this repo is not the right place to file such issues.

Thanks in advance,
ojas

**References**

[1] https://github.com/RPi-Distro/raspberrypi-sys-mods/tree/master/lib/udev/rules.d
[2] https://github.com/raspberrypi/userland/blob/master/interface/vchiq_arm/CMakeLists.txt#L9

XECDesign, 2021-09-23:
This should already be sorted. https://github.com/RPi-Distro/firmware/commit/cf341dbe355e4b5f60fbb150d61298c5e946547c

OjaswinM, 2021-09-25:
Ahh, never though it would be in a .patch file in the firmware repo haha. Anyways thanks for the update!

closed 2021-09-25 by XECDesign