RPi-Distro/repo #265: Dnsmasq ()package doesn't include dnsmasq binaries

msbroy, 2021-09-25:
Dnsmasq which was installed from raspbian repository with exact debian version: dnsmasq_2.80-1+rpt1+deb10u1_all.deb, doesn't include any dnsmasq binaries i.e. the /usr/sbin or /usr/bin directories were not packaged and pushed upstream. This is a serious issues. Maintainers please check this as soon as possible.

Attached image should the package structure: 

[image]

XECDesign, 2021-09-25:
Because they don't come from that package.

https://packages.debian.org/buster/all/dnsmasq/filelist
https://packages.debian.org/buster/armhf/dnsmasq-base/filelist

You're after dnsmasq-base.

msbroy, 2021-09-26:
Thanks for the explanation, and sorry for the confusion. :)

closed 2021-09-26 by msbroy