RPi-Distro/repo #266: No .Filesystem or .Encrypted interface on D-Bus object

beta-tester, 2021-10-08:
hi, there is an issue when i plug an external USB-SSD drive with encrypted content to the "Raspberry Pi OS Buster".

the partition is full encrypted by BitLocker.

when i plug the external USB-SSD to the Raspberry Pi, it pops up an error message:
`No .Filesystem or .Encrypted interface on D-Bus object`
also when i try to open the shown encrypted drive in PcManFM by double click on it, the same error message pops up again.
[image]

ENV:
- Raspberry Pi 4B (4GByte)
- original Raspberry Pi 4 USB-C Powersupply
- Raspberry Pi OS Buster (2021-05-07-raspios-buster-armhf.zip) + full-upgrade
- external USB-SSD, full encrypted by BitLocker

To verify i tried:
- [Debian 11 LXDE Live image](https://cdimage.debian.org/debian-cd/current-live/amd64/iso-hybrid/debian-live-11.0.0-amd64-lxde.iso) on PC
  -> no issues to access the external USB-SSD with full encrypted content BilLocker.
- [Ubuntu 21.04 Live image](https://releases.ubuntu.com/21.04/ubuntu-21.04-desktop-amd64.iso) on PC
  -> no issues to access the external USB-SSD with full encrypted content BilLocker.

because Debian distro and Ubuntu have not this issue, i guess it must be an issue specially related to Raspberry Pi OS.

beta-tester, 2021-11-28:
big progress with using RasPi OS Bullseye!

when plug in a BitlLocker encrypted partition now you get a Passphrase dialog instead of the Error above
[image]

but without installing the additional package `libblockdev-crypto2`, you will get an error message:
`Error unlocking /dev/sda1: The function 'bd_crypto_bitlk_open' called, but not implemented!`
[image]

and there are also errors, when you choose remembering the passphrase:
`Error storing passphrase in keyring (The name org.freedesktop.secrets was not provided by any .service files)`
[image]

and en error ewhen you try to eject the mounted drive again:
```
Failed to eject ...
Error ejecting /dev/sda: Command-line `eject '/dev/sda'` exited with non-zero exit status
1: eject unable to eject
```
[image]

XECDesign, 2021-11-29:
@spl237 Is this something that should be supported?

spl237, 2021-11-29:
It's not something we've ever tested or changed; any implementation is coming straight from upstream. So as far as I am concerned, we don't support it - if it doesn't work, blame Debian... ;)

XECDesign, 2021-11-29:
OP is saying it works in Debian running LXDE.

spl237, 2021-11-29:
In that case, pass - no idea why it would be working differently on us, as I have no idea where all the bitlocker stuff happens. I would guess that the dialogs above come from libfm, but none of the changes I have made should be affecting that.
The fact that an error message is talking about an unimplemented function suggests some library somewhere is missing - according to a quick Google, that function is in the libblockdev-crypto2 library, so should be present, unless something in the Raspbian build process is going wrong?
If I get some time at some point I can have a quick look, but it is really not going to be a priority, so it might be quicker for those in the community who care about this to investigate themselves in the meantime.

beta-tester, 2021-12-01:
according the keyring error message
`Error storing passphrase in keyring (The name org.freedesktop.secrets was not provided by any .service files)`
, the user **itsmedoofer** from the Raspberry Pi forum gave me the good hint to install `gnome-keyring` package.
and this solves that keyring error message.

PS.: not tested, but i think this is not BitLocker specific. other Linux specific filesystem encryption may have the same issue, at least that with the keyring and the one where the package `libblockdev-crypto2` needs to be installed.

XECDesign, 2021-12-01:
Sounds like we should probably include those two packages in the desktop image then.

libblockdev-crypto2 is a `recommends` of udisks2, so maybe that should be installed without `--no-recommends`.

`gnome-keyring` is usually dragged in as `recommends` by packages which use it directly or as a part of the desktop environment. Maybe it should be a recommends of the applet, and I can also add it explicitly to the image.

@spl237 thoughts?

beta-tester, 2021-12-01:
i played a bit with the excrypted device and logged syslog and mount output...
i found out i only get the error at eject, when using the eject option on the task bar.
when i use the eject button inside the File Manager (left top view where "Home Folder", "Filesystem Root" and my mounted encrypted device is listed) that will eject the encrypted device without any error messages in the log the first sequence (m1, m2, m3), the second sequence m11, m12, m13 i was using the eject button of the task bar and got again the error.

(maybe the File Manager was silently dumping the error)

```
pi@raspberrypi:~ $ mount &>m1.txt
plug in device
	Dec  1 04:19:05 raspberrypi kernel: [  101.875579] usb 1-1.5: new high-speed USB device number 5 using dwc_otg
	Dec  1 04:19:05 raspberrypi kernel: [  102.007458] usb 1-1.5: New USB device found, idVendor=0781, idProduct=55ae, bcdDevice=30.00
	Dec  1 04:19:05 raspberrypi kernel: [  102.007485] usb 1-1.5: New USB device strings: Mfr=2, Product=3, SerialNumber=1
	Dec  1 04:19:05 raspberrypi kernel: [  102.007499] usb 1-1.5: Product: Extreme 55AE
	Dec  1 04:19:05 raspberrypi kernel: [  102.007512] usb 1-1.5: Manufacturer: SanDisk
	Dec  1 04:19:05 raspberrypi kernel: [  102.007525] usb 1-1.5: SerialNumber: ...
	Dec  1 04:19:05 raspberrypi kernel: [  102.011662] usb 1-1.5: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
	Dec  1 04:19:05 raspberrypi kernel: [  102.011688] usb 1-1.5: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
	Dec  1 04:19:05 raspberrypi kernel: [  102.011703] usb-storage 1-1.5:1.0: USB Mass Storage device detected
	Dec  1 04:19:05 raspberrypi kernel: [  102.018492] scsi host0: usb-storage 1-1.5:1.0
	Dec  1 04:19:05 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
	Dec  1 04:19:05 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
	Dec  1 04:19:05 raspberrypi kernel: [  102.094433] usbcore: registered new interface driver uas
	Dec  1 04:19:06 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
	Dec  1 04:19:06 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
	Dec  1 04:19:08 raspberrypi kernel: [  104.104259] scsi 0:0:0:0: Direct-Access     SanDisk  Extreme 55AE     3000 PQ: 0 ANSI: 6
	Dec  1 04:19:08 raspberrypi kernel: [  104.106751] sd 0:0:0:0: [sda] 3906963617 512-byte logical blocks: (2.00 TB/1.82 TiB)
	Dec  1 04:19:08 raspberrypi kernel: [  104.112618] scsi 0:0:0:1: Enclosure         SanDisk  SES Device       3000 PQ: 0 ANSI: 6
	Dec  1 04:19:08 raspberrypi kernel: [  104.113236] sd 0:0:0:0: [sda] Write Protect is off
	Dec  1 04:19:08 raspberrypi kernel: [  104.113262] sd 0:0:0:0: [sda] Mode Sense: 17 00 10 00
	Dec  1 04:19:08 raspberrypi kernel: [  104.116832] sd 0:0:0:0: [sda] No Caching mode page found
	Dec  1 04:19:08 raspberrypi kernel: [  104.116869] sd 0:0:0:0: [sda] Assuming drive cache: write through
	Dec  1 04:19:08 raspberrypi kernel: [  104.148363] sd 0:0:0:0: Attached scsi generic sg0 type 0
	Dec  1 04:19:08 raspberrypi kernel: [  104.148579] scsi 0:0:0:1: Attached scsi generic sg1 type 13
	Dec  1 04:19:08 raspberrypi kernel: [  104.180201]  sda: sda1
	Dec  1 04:19:08 raspberrypi kernel: [  104.183710] sd 0:0:0:0: [sda] Attached SCSI disk
	Dec  1 04:19:08 raspberrypi dbus-daemon[597]: [session uid=1000 pid=597] Activating service name='org.freedesktop.secrets' requested by ':1.12' (uid=1000 pid=831 comm="/usr/libexec/gvfs-udisks2-volume-monitor ")
	Dec  1 04:19:10 raspberrypi gnome-keyring-d[1172]: couldn't access control socket: /run/user/1000/keyring/control: No such file or directory
	Dec  1 04:19:10 raspberrypi dbus-daemon[597]: [session uid=1000 pid=597] Successfully activated service 'org.freedesktop.secrets'

unlock (with remember password until logout) & mount
	Dec  1 04:19:27 raspberrypi kernel: [  123.320451] device-mapper: ioctl: 4.43.0-ioctl (2020-10-01) initialised: dm-devel@redhat.com
	Dec  1 04:19:30 raspberrypi kernel: [  126.193543] cryptd: max_cpu_qlen set to 1000
	Dec  1 04:19:30 raspberrypi udisksd[421]: Unlocked device /dev/sda1 as /dev/dm-0
	Dec  1 04:19:32 raspberrypi ntfs-3g[1224]: Version 2017.3.23AR.3 integrated FUSE 28
	Dec  1 04:19:32 raspberrypi ntfs-3g[1224]: Mounted /dev/mapper/bitlk-2049 (Read-Write, label "USB-SSD#1", NTFS 3.1)
	Dec  1 04:19:32 raspberrypi ntfs-3g[1224]: Cmdline options: rw,nodev,nosuid,uid=1000,gid=1000,windows_names,uhelper=udisks2
	Dec  1 04:19:32 raspberrypi udisksd[421]: Mounted /dev/dm-0 at /media/pi/USB-SSD#1 on behalf of uid 1000
	Dec  1 04:19:32 raspberrypi ntfs-3g[1224]: Mount options: nodev,nosuid,uhelper=udisks2,allow_other,nonempty,relatime,rw,default_permissions,fsname=/dev/mapper/bitlk-2049,blkdev,blksize=4096
	Dec  1 04:19:32 raspberrypi ntfs-3g[1224]: Global ownership and permissions enforced, configuration type 7
	Dec  1 04:19:32 raspberrypi dbus-daemon[597]: [session uid=1000 pid=597] Activating via systemd: service name='org.gtk.vfs.Metadata' unit='gvfs-metadata.service' requested by ':1.10' (uid=1000 pid=698 comm="pcmanfm --desktop --profile LXDE-pi ")
	Dec  1 04:19:32 raspberrypi systemd[543]: Starting Virtual filesystem metadata service...
	Dec  1 04:19:32 raspberrypi dbus-daemon[597]: [session uid=1000 pid=597] Successfully activated service 'org.gtk.vfs.Metadata'
	Dec  1 04:19:32 raspberrypi systemd[543]: Started Virtual filesystem metadata service.

[... cut ...]
	Dec  1 04:26:21 raspberrypi udisksd[421]: Locked device /dev/sda1 (was unlocked as /dev/dm-0)
	Dec  1 04:26:21 raspberrypi kernel: [  537.254630]  sda: sda1

pi@raspberrypi:~ $ mount &>m13.txt
pi@raspberrypi:~ $ diff m12.txt m13.txt 
22d21
< /dev/mapper/bitlk-2049 on /media/pi/USB-SSD#1 type fuseblk (rw,nosuid,nodev,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096,uhelper=udisks2)

	Dec  1 04:28:23 raspberrypi kernel: [  659.405156] usb 1-1.5: USB disconnect, device number 6

```

beta-tester, 2021-12-02:
i also took a look to the output of `sudo journalctl -xeu udisk*` and there it shows some errors, even there is no encrypted device or raid plugged in (it is about `crypto`, `mdraid` and `smart`):
```
Dec 01 16:10:25 server systemd[1]: Starting Disk Manager...
░░ Subject: A start job for unit udisks2.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit udisks2.service has begun execution.
░░ 
░░ The job identifier is 163.
Dec 01 16:10:25 server udisksd[516]: udisks daemon version 2.9.2 starting
Dec 01 16:10:25 server udisksd[516]: failed to load module crypto: libbd_crypto.so.2: cannot open shared object file: No such file or directory
Dec 01 16:10:25 server udisksd[516]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory
Dec 01 16:10:25 server udisksd[516]: Failed to load the 'mdraid' libblockdev plugin
Dec 01 16:10:25 server udisksd[516]: Failed to load the 'crypto' libblockdev plugin
Dec 01 16:10:27 server systemd[1]: Started Disk Manager.
░░ Subject: A start job for unit udisks2.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit udisks2.service has finished successfully.
░░ 
░░ The job identifier is 163.
Dec 01 16:10:27 server udisksd[516]: Acquired the name org.freedesktop.UDisks2 on the system message bus
Dec 01 18:30:49 server udisksd[516]: Error performing housekeeping for drive /org/freedesktop/UDisks2/drives/portable_SSD_...: Error updating SMART data: sk_disk_smart_read_data: Operation not supported (udisks-error-quark, 0)
Dec 01 19:01:21 server udisksd[516]: Error performing housekeeping for drive /org/freedesktop/UDisks2/drives/portable_SSD_...: Error updating SMART data: sk_disk_smart_read_data: Input/output error (udisks-error-quark, 0)

```

beta-tester, 2021-12-04:
i checked again some logs...
the above message:
`Dec 01 16:10:25 server udisksd[516]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory`
i also get on Debian 11.1 LXDE Live.
but not the others.

XECDesign, 2022-01-11:
I don't have appropriate devices to test with, but once the next image is release it would be interesting to hear whether it improves the situation. It should include a few of the packages you mentioned and some others, which may help, but there's no ETA yet.

beta-tester, 2022-01-11:
i will try it out as soon as i see the new released image and give you feedback.

beta-tester, 2022-01-12:
> ... once the next image is release ... it ... include a few of the packages ... and some others

### OFFTOPIC
the packages `zip` and `p7zip` would be nice to have included as well
to give the `Xarchiver` application the ability to open and create `.zip` and `.7z` archives out of the box.
but it is only a suggestion - `zip` and `7z` i use very often coming from windows environmnts.

XECDesign, 2022-01-12:
Yup, good call.

beta-tester, 2022-01-30:
hi, i just tried the new release image "2022-01-28-raspios-bullseye-armhf.zip"
- i can mount/access encrypted volumes by entering the password,
- i can remember the password.
- i have `7zip` and `zip` archives available in XArchive.

nice - thank you very much! :smile:

what still is odd:
- ejecting the encrypted volume by using the eject button on the taskbar throws still an errormessage:
  [image]
  ```
  Failed to eject ...
  Error ejecting /dev/sda: Command-line `eject
  '/dev/sda'` exited with non-zero exit status
  1: eject unable to eject
  ```

XECDesign, 2022-02-01:
Great, thanks for testing!

And you can still eject from pcmanfm without any issues?

beta-tester, 2022-02-01:
yes... at least it gives no error like the eject of the taskbar does.

when device plugged, unlocked with password and mounted it shows like this in pcmanfm:
[image]

when i click to the eject icon in pcmanfm, pcmanfm closes.
when i open pcmanfm again i can see the unmounted encrypted device like this:
[image]

so pcmanfm does not eject that device, it only unmount it.

then i again test the behavior with Debian LXDE on a PC. it behaves the same.

but Debian LXDE does not have an eject button on the taskbar so i could not see how it behaves with that.

i did some tests on the RPi on the console and it seems the /dev/sda is in use by some other process, but i can't figure out which process it is. `sudo lsof /dev/sda` and `sudo fuser /dev/sda` do not give any output...

when the encrypter device is plugged, unlocked by password and mounted:
```bash
pi@test:~ $ sudo lsblk
NAME           MAJ:MIN RM  SIZE RO TYPE  MOUNTPOINT
sda              8:0    0  1,8T  0 disk  
└─sda1           8:1    0  1,8T  0 part  
  └─bitlk-2049 254:0    0  1,8T  0 crypt /media/pi/USB-SSD#1
mmcblk0        179:0    0 29,7G  0 disk  
├─mmcblk0p1    179:1    0  256M  0 part  /boot
└─mmcblk0p2    179:2    0 29,5G  0 part  /

pi@test:~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=340904k,nr_inodes=85226,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=188996k,nr_inodes=819200,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94496k,nr_inodes=23624,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/mapper/bitlk-2049 on /media/pi/USB-SSD#1 type fuseblk (rw,nosuid,nodev,relatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096,uhelper=udisks2)
```

after i clicked the eject button in pcmanfm:
```bash
pi@test:~ $ sudo lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda           8:0    0  1,8T  0 disk 
└─sda1        8:1    0  1,8T  0 part 
mmcblk0     179:0    0 29,7G  0 disk 
├─mmcblk0p1 179:1    0  256M  0 part /boot
└─mmcblk0p2 179:2    0 29,5G  0 part /

pi@test:~ $ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=340904k,nr_inodes=85226,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
[... 27 lines cut ...]
├─mmcblk0p1    179:1    0  256M  0 part  /boot
└─mmcblk0p2    179:2    0 29,5G  0 part  /

pi@test:~ $ sudo eject /dev/sda
eject: cannot open /dev/sda: Das Gerät oder die Ressource ist belegt
pi@test:~ $ sudo eject /dev/sda1
eject: cannot open /dev/sda: Das Gerät oder die Ressource ist belegt
pi@test:~ $ sudo eject /dev/mapper/bitlk-2049 
eject: /dev/mapper/bitlk-2049: is not hot-pluggable device
pi@test:~ $ sudo eject /media/pi/USB-SSD#1/
eject: /dev/mapper/bitlk-2049: is not hot-pluggable device
pi@test:~ $ sudo umount /media/pi/USB-SSD#1 
pi@test:~ $ sudo eject /dev/mapper/bitlk-2049 
eject: /dev/mapper/bitlk-2049: is not hot-pluggable device
pi@test:~ $ sudo eject /dev/sda1
eject: cannot open /dev/sda: Das Gerät oder die Ressource ist belegt
pi@test:~ $ sudo eject /dev/sda
eject: cannot open /dev/sda: Das Gerät oder die Ressource ist belegt
```

is the Ejecter on the taskbar a custom application?
if so, maybe it would be good to list only "hot-pluggable" devices.
or if trying to eject a non-hot-pluggable device, do only unmount it without sowing an error message.

if it is an app from Debian, can you tell me where to find it to test under Debian LXDE Live.

XECDesign, 2022-02-01:
I believe it's custom.

@spl237 Looks like it runs `eject` in cases when it doesn't make sense to do so? AIUI, it should umountm, undo any loopback setup that was done (I guess udisks2 would handle all that) and then eject the actual hardware path like sda rather than trying to eject the mapper path.

XECDesign, 2022-02-17:
Thanks again for investigating the issue and providing the fixes. The remaining issue with the ejecter can be reported here: https://github.com/raspberrypi-ui/lxplug-ejecter/issues

closed 2022-02-17 by XECDesign