RPi-Distro/repo #269: [Bullseye+Buster] missing 'mage' package crashes x session

jbarth-ubhd, 2021-11-10:
... when configured.

After installing *mage* with `apt install mage` everything works fine.

Package is not by default in *Bullseye*, even in *Raspberry Pi OS with desktop and recommended software*

XECDesign, 2021-11-10:
@spl237

spl237, 2021-11-10:
mage should not be installed by default. It should only be installed when required; adding the Magnifier using Recommended Software installs it. The lack of it does not crash the X session on any bullseye image I have used in several months of testing.

What are you actually doing when the X session crashes?

jbarth-ubhd, 2021-11-10:
When trying to configure the Not installed Magnifier(Right mouse button on taskbar), x Session crashes.

spl237, 2021-11-10:
Do you mean if you right-click the taskbar, open Panel Preferences, go into Panel Applets, select the Magnifier and then click the Preferences button?

I can indeed crash the X session like that - which I will look into fixing - but I can't find any other way of causing a crash, and that sequence is pretty unlikely to happen very often!

spl237, 2021-11-10:
Fixed crash when that button is pressed - if you find any other way of causing the crash, please re-open the issue.

Many thanks!

closed 2021-11-10 by spl237