RPi-Distro/repo #270: nodered depends on python (>= 2.7) leading to inconsistency between "with desktop" and "full" images.

plugwash, 2021-11-11:
The pi-gen scripts explicitly install python-is-python3 which makes /usr/bin/python point to python3, this is the sensible modern choice and should if-possible be retained.

Unfortunately nodered depends on "python (>= 2.7)" on bullseye this dependency is satisfied by the python-is-python2 package which conflict with the python-is-python3 package.

This results in the "with desktop" and "with desktop and recommended software" images having /usr/bin/python pointing at different versions of python which is a pretty bad state of affairs

IMO nodered needs to be fixed, ideally to use python3, but if that is not possible then at least fixed to use python2 explicitly instead of via the /usr/bin/python symlink and depend on "python2" instead of "python". 

Unfortunately I notice that the nodered binary package does not appear to have a corresponding source package, so it would be difficult for me to prepare a fix. I also notice that the nodered package seems to be a rather old version, should it really be shipped as part of a default install if it's not going to be updated?

XECDesign, 2021-11-12:
@spl237

spl237, 2021-11-12:
Raised with IBM.

dceejay, 2021-11-12:
Happy for you to pull the package from Bullseye repo, while we work on it.
As usual the default debian node.js version is still behind the current LTS (12 vs 14) - so that's not ideal either.

plugwash, 2021-12-01:
I don't see any reason to pull it from the repo, but I do think it should be pulled from the image build, so we get a consistent "python3 as default" position out of the box.

dceejay, 2021-12-01:
We are about to release a fixed version that builds correctly and just uses python3 - we will send the updated version as soon as we have released and re-rested the build.

dceejay, 2021-12-02:
Hi - I have sent Simon the latest deb for Bullseye repo ONLY 
as it pre-reqs Python3 and the later nodejs/npm versions.

XECDesign, 2021-12-02:
Many thanks Dave, these should be live now.

dceejay, 2021-12-02:
Thanks, hopefully @plugwash can check and close this issue .

plugwash, 2021-12-02:
I have downloaded the new deb and confirmed that it no longer depends on the "python" package.

the next step would be to run a new image build and confirm that the "with desktop and reccomended software" image now ends up with python-is-python3 installed as expected.

XECDesign, 2022-02-16:
Both images now come with python-is-python3:

```
>  curl -s http://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf-full.info | grep python-is-
ii  python-is-python3                     3.9.2-1                           all          symlinks /usr/bin/python to python3
> curl -s http://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf.info | grep python-is-
ii  python-is-python3                    3.9.2-1                          all          symlinks /usr/bin/python to python3
```

Many thanks

closed 2022-02-16 by XECDesign